elden1337/hass-peaq #7: Add Garo

elden1337, 2022-04-09:


elden1337, 2022-04-11:
https://github.com/sockless-coding/garo_wallbox/tree/master/custom_components/garo_wallbox

stejo388, 2023-01-10:
Hi, just wondering about the status regarding the GARO integration?

elden1337, 2023-01-10:
Hi! 
I havent heard from anyone trying to set it up (i dont have one myself so have to rely on others). 
Did you attempt to install it?

joek-makewave, 2023-05-25:
Hi there,
  I have the Garo wallbox (model GLBDCWM-T222WO). I saw that support for it has been added in a recent release of hass-peaq and tried an install using the main branch.

I quickly found some errors in garowallbox.py regarding entity names that was easy to correct. That got rid of some errors at start-up. The first problem in the log is now:

```
2023-05-25 23:40:08.078 WARNING (MainThread) [custom_components.peaqev.peaqservice.chargertypes.types.garowallbox] You are initiating GaroWallbox as Chargertype. Bare in mind that this chargertype has not been signed off in testing and may be very unstable. Report findings to the developer.
<peaqevcore.services.locale.Locale.LocaleData object at 0x11bdf84f0>
[]
2023-05-25 23:40:08.449 ERROR (MainThread) [homeassistant.components.filter.sensor] Could not convert state:  (<class 'str'>) to number
2023-05-25 23:40:08.451 ERROR (MainThread) [homeassistant.components.filter.sensor] Could not convert state:  (<class 'str'>) to number
```
A bit later I have:
```
2023-05-25 23:40:12.390 WARNING (MainThread) [peaqevcore.models.hub.chargerswitch] Chargerobject state was None while getting current. Updatecurrent will still process but you may see a missmatch in frontend.
2023-05-25 23:41:07.956 ERROR (MainThread) [homeassistant.helpers.event] Error while processing state change for sensor.peaqev_energy_including_car
Traceback (most recent call last):
  File "/Users/jonas/git/hass-peaq/venv/lib/python3.10/site-packages/homeassistant/helpers/event.py", line 288, in _async_state_change_dispatcher
    hass.async_run_hass_job(job, event)
  File "/Users/jonas/git/hass-peaq/venv/lib/python3.10/site-packages/homeassistant/core.py", line 577, in async_run_hass_job
    hassjob.target(*args)
  File "/Users/jonas/git/hass-peaq/venv/lib/python3.10/site-packages/homeassistant/components/utility_meter/sensor.py", line 366, in async_reading
    for sensor in self.hass.data[DATA_UTILITY][self._parent_meter][
KeyError: 'utility_meter_data'```

And the state for sensor "Peaqev Charger controller" goes from Stop to Error.

I'm interested in continuing with this. What do you think is the best way to proceed?

elden1337, 2023-05-25:
Great, yeah I need testers for Garo ;). 

Ill have a look and get back with how to proceed!

elden1337, 2023-05-26:
2023-05-25 23:40:08.449 ERROR (MainThread) [homeassistant.components.filter.sensor] Could not convert state:  (<class 'str'>) to number

This particular log I think is unrelated to Garo. What do you use as power meter sensor? I don't think i'm floatparsing before putting it into the average-sensors. So i'll double check that.

elden1337, 2023-05-26:
Which ones of all sensors for peaqev are unavailable for you after boot?

elden1337, 2023-05-27:
@joek-makewave are these garo-states actually written in caps, or are they lower-case?

```python
    def _set_charger_states(self) -> None:
        self.chargerstates[ChargeControllerStates.Idle] = ["NOT_CONNECTED"]
        self.chargerstates[ChargeControllerStates.Connected] = [
            "CONNECTED",
            "CHARGING_PAUSED",
            "CHARGING_CANCELLED",
        ]
        self.chargerstates[ChargeControllerStates.Done] = ["CHARGING_FINISHED"]
        self.chargerstates[ChargeControllerStates.Charging] = ["CHARGING"]
```

joek-makewave, 2023-05-29:
> @joek-makewave are these garo-states actually written in caps, or are they lower-case?
>
> ```python
>     def _set_charger_states(self) -> None:
>         self.chargerstates[ChargeControllerStates.Idle] = [...]

They are in lower case.
"Vehicle not connected"
"Vehicle connected"
"Charging paused"
"Charging cancelled"
"Charging finished"
"Charging"

joek-makewave, 2023-05-29:
> 

> 2023-05-25 23:40:08.449 ERROR (MainThread) [homeassistant.components.filter.sensor] Could not convert state: (<class 'str'>) to number
>
> This particular log I think is unrelated to Garo. What do you [...]

OK. I use the load balancer of the Garo wallbox. The power sensor seems to be ok. It works as expected in other places. But I think I might have spotted an error in the Garo wallbox integration. It looks like some of the sensors in the garo wallbox integration with string state cast to int in `native_value`.

joek-makewave, 2023-05-29:
> Which ones of all sensors for peaqev are unavailable for you after boot?

[image]

joek-makewave, 2023-05-29:
I made a pull request with the changes I did for the sensor names. There is also a fix for the values of the set mode service call.

closed 2024-10-26 by elden1337