elden1337/hass-peaq #16: Chargepoint ID

nsimb, 2022-04-18:
I feel really stupid but i cant get it to work, my guess is that i give the wrong Chargepoint ID
Tried all kind of stuff related to device and entitys

Checked out the code to see what it could be related to and tested. Sadly no luck

Easee charger, device is called 42 but i tried with different entitys aswell without luck
I asume that the power sensor should the the full entity name and mesure in W since power. Have a few but so far no luck

[image]

elden1337, 2022-04-18:
It s a bug in the config validation. Think i just patched it in latest release.

nsimb, 2022-04-18:
works fine thanks

closed 2022-04-18 by nsimb

elden1337, 2022-04-18:
Make sure to update to latest. 
Have had a dance with latest HA tonight and the previous releases dont work in the latest HA.