elden1337/hass-peaq #18: Setup failes with error on energy sensor

nsimb, 2022-04-19:
I get different errors with different sensors, see log below.
My asumption is that eithere the naming, or attributes on them do not match up.
Or simply that i have miss judge what sensors that should be provided.

[image]

At setup it accepts my input and configuration steps. But then gives error that it cant be set up.
Full log below with modified entity due to adress (for the Tibber Pulse)
First i thought it was about that i have local production so some sensors goes negative, however tested during over consuption with same error.

Let me know what data i can collect and what assistans i can give/help out with

2022-04-19 10:16:08 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry sensor.total_power_total_mqtt for peaqev
File "/config/custom_components/peaqev/__init__.py", line 32, in async_setup_entry
File "/config/custom_components/peaqev/peaqservice/hub.py", line 105, in __init__
File "/config/custom_components/peaqev/peaqservice/hub.py", line 133, in init_hub_values
File "/config/custom_components/peaqev/peaqservice/util/hubmember.py", line 30, in value
2022-04-19 10:20:02 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry sensor.power_XXXadressXXX for peaqev
File "/config/custom_components/peaqev/__init__.py", line 32, in async_setup_entry
File "/config/custom_components/peaqev/peaqservice/hub.py", line 80, in __init__
File "/config/custom_components/peaqev/config_flow.py", line 77, in async_step_user
File "/config/custom_components/peaqev/config_flow.py", line 56, in validate_input_user
File "/config/custom_components/peaqev/config_flow.py", line 38, in _check_power_sensor
2022-04-19 10:31:51 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry sensor.power_XXXadressXXX for peaqev
File "/config/custom_components/peaqev/__init__.py", line 32, in async_setup_entry
File "/config/custom_components/peaqev/peaqservice/hub.py", line 80, in __init__

elden1337, 2022-04-19:
Hi, I did a lot of changes yesterday evening with another guy using easee. Make sure you are on latest release that I just pushed ([https://github.com/elden1337/hass-peaq/releases/tag/v0.3.4](https://github.com/elden1337/hass-peaq/releases/tag/v0.3.4)) and see if it works there. I guess this could be the sensors you have here seem to be correct (taken from what they are named at least).

Since I haven't been playing around with negative values (solar panels?) there will most likely be some issues there that we'll have to attend to.

elden1337, 2022-04-19:
seems from just glancing here that the "sensor.total_power_total_mqtt" is passing it's own setup, but that the init in hub has issues with (what i guess is) max-current of your easee. Let's see what happens with the new version. 

that "sensor.power_XXXadressXXX", does it ever pass the config flow part? I think it has troubles reading the state of that sensor when testing the config.

nsimb, 2022-04-19:
relized that I hade one sub version to low off home assistant and the latest version was a requirement in HACS for the latest version. So I was one version off (yeasterdays update) sorry. There for it did not show up as avalible.
Fixed this and after both updates the integrations started. Right now Iam using a sensor that does not go negative. Will do a few more tests with different sensors and post back. Might only be that I missed latest version due to I was behind on HA subversion.
//nsimb

elden1337, 2022-04-19:
Well,now comes the funny part. To see exactly how badly i blind coded easee type since i cant test it myself. 

Looking forward to your findings!

nsimb, 2022-04-19:
Tried with a few different sensors and I relize that i get a error when setting up the integration with all sensors. But if reboot after that it works anyway. No big problem, but could be good to know.

Looks like all entities in the integration did get values and keeps getting updated so far so good.

Since iam overproducing atm i have some trouble testing atm. Will check later tonight when the sun is down

elden1337, 2022-04-19:
Fantastic. Is the chargercontroller-state something other than error then as well?

nsimb, 2022-04-19:
Right now its "Unknown" atm that does not sound to good. However I keep the charger disable atm until tonight to have fewer parameters for tests.

elden1337, 2022-04-19:
[image]

the one on top is always unknown. The highlighted one i'd expect to be "error", "idle", "connected" or possibly "done" in yours. But let's revisit tonight.

nsimb, 2022-04-20:
last night test did go great, everything registered and updated, including charger controller
The solution did not need to do anything to avoid Peak increase.

So next step is to change configuration and the amount that needs to be charged so the solution does react, need to do some preperations for that. Will keep you updated. Off tonight so will be later this week sadly

elden1337, 2022-04-20:
Awesome news mate! 
Keep me posted. We'll keep this ticket open til you've satisfied all your testing and then you can close it once confident. 

Thanks for testing this!

closed 2022-04-27 by nsimb