elden1337/hass-peaq #33: Add Zaptec

elden1337, 2022-04-23:


elden1337, 2022-05-27:
https://github.com/custom-components/zaptec

gustafssone, 2022-06-12:
I have a zaptec go, let me know if I can help out for testing, etc. Using that mentioned integration.

elden1337, 2022-06-17:
@gustafssone i've started this now. 
Can you assist me with:

* It doesn't seem as if one can update the maxcurrent through this integration. is that correct? if so, we can still integrate it, but the experience will be less ideal.
* Can you provide me a list of sensors from this integration so that i know what i need to add?

elden1337, 2022-06-17:
you got an alpha-version. I'm not sure it will start, but would like to see what happens when you install it.

What is expected is that it installs, and that start/stop/pause/resume will work. It should also be able to read current power draw (i hope). 
I have not found any possibilities of setting amps, if you know how to see them please give insight.

https://github.com/elden1337/hass-peaq/releases/tag/v0.14.0-alpha

gustafssone, 2022-06-20:
Appreciate the fast response.

@elden1337 have a look at "Update charger properties (requires owner or service permissions)." at https://api.zaptec.com/help/index.html#/ . That sets the current max charge right? Because I believe that POST command isn't implemented. I'm not too familiar with  python (more JavaScript), but I might be able to add a change request to that repository.

The power sensor is an attribute on the sensor, e.g. sensor.zaptec_charger_zap034545 . They made some examples at https://github.com/custom-components/zaptec/tree/master/lovelace_example , so mine looks like this:

`    zaptec_01_power:
      entity_id: sensor.zaptec_charger_zap034545
      friendly_name: "Car charger power"
      icon_template: mdi:power-plug
      unit_of_measurement: "W"
      value_template: "{{ state_attr('sensor.zaptec_charger_zap034545', 'total_charge_power') | round(0) }}"`

Do you want the charging feature to be as a switch? Based on the current standard implementation it is a service call, but I can just add a switch template.

Also, the charging doesn't work with "Permission required" (it uses RFID), so let me just add a webhook to confirm I can start it when I want. Right now it starts immediately when adding cable to car. 

[image]

gustafssone, 2022-06-22:
@elden1337 well, I got the webhook to work! https://github.com/gustafssone/zaptecwebhook 
Now the charging waits until authorization. I will see if I can limit amps - not too familiar with python but might get it to work.

gustafssone, 2022-06-22:
Hmm, it looks like it is this one: https://api.zaptec.com/help/index.html#/Installation/post_api_installation__id__update . And there is code prepared, but not sure how to use it (it doesn't exist as a service): https://github.com/custom-components/zaptec/blob/b5b487062ee051f3d8e9b8c85afacfbcc4b62d6e/custom_components/zaptec/api.py#L168

gustafssone, 2022-06-23:
Great. I have a discussion with the developer there, so might solve it myself.

elden1337, 2022-07-03:
awaiting this pr and update of zaptec to continue: https://github.com/custom-components/zaptec/pull/23

@gustafssone i'll assign you to this for now.

gustafssone, 2022-07-05:
Yep, sounds good. Got stuck at changing charge atm, but shouldn't be too hard to figure out.

closed 2023-07-15 by elden1337