elden1337/hass-peaq #49: Cant set up the Peaq in integration, still getting Aborted

nsimb, 2022-05-03:
closed the last ticket, after clean install, and trying to figure out what could be wrong with my setup.
[image]

Guessing it got somthing with how to with the additional of the Nordpool, since this started to hapen when the priceaware beta got released.

Not really sure how to debugg or find the logs when it comes to custom integration setup throu gui. Let me know if its just my installation and what i can do.

v0.5.3 
core-2022.4.7
supervisor-2022.04.0
Home Assistant OS 7.6

elden1337, 2022-05-03:
hmm, weird. the nordpool is supposed to be optional ([manifest](https://github.com/elden1337/hass-peaq/blob/main/custom_components/peaqev/manifest.json)) like the charger-types (i don't have easee for example). 

I guess, that something's lingering in your config-files which for some reason is corrupt. The installation is then trying to add to that again. 

Is the "Aborted" happening straight away or on posting the parameters?

2022.4.5 is what is required (but cannot enforce that for other reasons) but you seem to have it covered version-wise. 

However, i managed to reproduce your issue here now with the latest. will check what's up and get back.

nsimb, 2022-05-03:
It happens straight away, before any othere screen is displayed. Not giving any options for doing things differently
Thanks for checking this out

elden1337, 2022-05-03:
Yepp. Noticed as well. 
I must have crapped it when adding options to it, so debugging now

elden1337, 2022-05-03:
At first i thought it was that your instance had kept the old config. But i get the same error and my config was deleted when i removed the integration.

elden1337, 2022-05-03:
There we go. harder than i thought. Had to remove the options flow for now, but this is now tested several times for me. 
https://github.com/elden1337/hass-peaq/releases/tag/v0.5.4

closed 2022-05-03 by elden1337

nsimb, 2022-05-04:
did work fine now to set it up, looks to be some transaltion or descriptions since setup did not show text for the fields in the secund view of the setup (houres for charging and optimice and checkbox) but that does not mater for the beta testing. Will play around and test abit
thanks

elden1337, 2022-05-04:
@nsimb https://github.com/elden1337/hass-peaq/releases/tag/v0.5.5