elden1337/hass-peaq #60: Add Belgium

djfanatix, 2022-05-17:
Could you add Belgium in the future?

Peak will be calculated for a period of 15 minutes.

The average of 12 month peaks will be the chargeable peak.

elden1337, 2022-05-17:
Absolutely. Is this a national model? 
Can the 12 peaks be on the same day or do they have to be on different days?

elden1337, 2022-05-17:
Or is there a link to docs? I can read some dutch and some french

djfanatix, 2022-05-17:
[](https://www.vreg.be/nl/nieuwe-nettarieven)

Here is some info.

No, so every 15 minutes the peak is calculated (to level out the instant high peaks)

The highest peak of a month is then the 'month peak'

The average of 12 months peaks (over a year) is the peak which you are billed for.

elden1337, 2022-05-17:
Ok nice. So in the app i then probably only have to care about the monthly registered (since i only deal with pushing down the quarters of each hour). 

Also an interesting case since id rather not have the ev chargers jump in and out too much, so will lab with it.

elden1337, 2022-05-19:
you don't have nordpool as an area in belgium. What integration would you use in your HA to see the current hourly energy price?

djfanatix, 2022-05-19:
Yes, I use Noordpool with sensor: sensor.nordpool_kwh_be_eur

https://www.nordpoolgroup.com/en/Market-data1/Dayahead/Area-Prices/be/hourly/?view=table

elden1337, 2022-05-19:
also, i'm gonna name this locale-type Belgium (VREG) since it seems to be mostly for Vlanders when i read through the page? Ie there could be other energy companies doing different ones for other regions?

djfanatix, 2022-05-19:
The other regions I don't know at this moment, It is indeed only for Flanders.
There are 2 energy companies providing this service. (Engie and Eneco)

elden1337, 2022-05-19:
Splendid. We'll go from there then. I will push a beta release for it (without any charged peak shown since that is a bit gnarly with the statistics i think (store a whole year and average it monthly, but then what happens with the current year? do i just average the months i have and cycle at new year.. etc). 

It won't affect peaqev in any way so i think i might drop that one in the beta and we'll see where we get.

elden1337, 2022-05-19:
Here you go mate @djfanatix : https://github.com/elden1337/hass-peaq/releases/tag/v0.7.0-beta

Please verify when you can.

djfanatix, 2022-05-20:
Thanks!, I installed it, I have to wait for less sun to see the peaks from import :-)
I'll keep you updated

elden1337, 2022-05-23:
@djfanatix have you seen any power-readings from the integration yet? If so, I think it should be fine all of it. 
I've been running the code on my machine but with the legacy-setup (through the new split logic then ofc) and it works.

djfanatix, 2022-05-23:
Yeah looks good to me, I couldn't really test the peaks yet, as my battery is still full.
But I didn't noticed any bugs

elden1337, 2022-05-23:
Ill merge it in general release. We ll hotfix if things emerge

closed 2022-05-23 by elden1337