elden1337/hass-peaq #68: QA Easee type

elden1337, 2022-05-28:
Help wanted. Someone with easee needs to qa it since im having a hard time finalizing without the possibility to debug.

reinestenberg, 2022-06-02:
I have an Easee and embedded SW skills (C). Python and home assistant API is quite limited

elden1337, 2022-06-02:
hi @reinestenberg, fantastic! 

if you install the component and reboot HA, you should see stuff in logs. 
My concerns are mainly:

the chargerentity. According to what i can find, that entity should end with _status for easee. But for other installations with easee, I get errors because if reports no state or state 0. So i'm unsure if that sensor is the correct one. 

the other thing is that on/off. For easee, it's not a switch as it seems, but rather turn the allowed amp up and down. This part i think i've gotten correctly, but have yet to see it working on an installation.

elden1337, 2022-06-02:
Yes, if you do modify reboot is needed. 

But was this in the initial configuration? 
What kind of sensor did you put as powersensor?

reinestenberg, 2022-06-02:
Occurs in the downloadet code. By hacs.
I just wanted to try some things to pinpoint the error.

The sensor is my own sensor implemented in appdeamon
But it does not matter what I input in the sensor field. I get unexpected
all the time. I have not tested a invalid sensor right after a reboot
however.

My own sensor is visible in developer tools/template and gets a nice float
there. I copy paste the name from there to be sure.

The unexpected error occurs when I have entered göteborg energi and
selected easee and pressing next. Not in the sensor dialog.

reinestenberg, 2022-06-02:
Found this log:
Logger: custom_components.peaqev.config_flow
Source: custom_components/peaqev/config_flow.py:79
Integration: peaqev ev-charging ([documentation](https://github.com/elden1337/hass-peaq), [issues](https://github.com/elden1337/hass-peaq/issues))
First occurred: 19:49:03 (4 occurrences)
Last logged: 20:11:38
Unexpected exception

Traceback (most recent call last):
  File "/config/custom_components/peaqev/config_flow.py", line 79, in async_step_charger
    await ConfigFlowValidation.validate_power_sensor(self.hass, user_input["powersensor"])
KeyError: 'powersensor'

elden1337, 2022-06-02:
good catch. i did break something... pushing fix and deploying new version now. Thanks

elden1337, 2022-06-02:
https://github.com/elden1337/hass-peaq/releases/tag/v0.10.1

reinestenberg, 2022-06-02:
Logger: homeassistant.config_entries
Source: custom_components/peaqev/__init__.py:49
Integration: peaqev ev-charging ([documentation](https://github.com/elden1337/hass-peaq), [issues](https://github.com/elden1337/hass-peaq/issues))
First occurred: 21:03:28 (1 occurrences)
Last logged: 21:03:28
Error setting up entry sensor.pws_power_sum for peaqev

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 335, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/peaqev/__init__.py", line 49, in async_setup_entry
    ci["cautionhours"] = conf.options["cautionhours"] if "cautionhours" in conf.options.keys() else conf.data["cautionhours"]
KeyError: 'cautionhours'

elden1337, 2022-06-02:
ok, more corner case but still good catch. 
I guess you chose to not go with price-aware, and then in nonhours and cautionhours didn't select any caution-hours?

For speed purposes, if you'd like to replace your __init__.py with this code it should create empty lists if not existing:

```
from __future__ import annotations

import logging

from homeassistant.config_entries import ConfigEntry
from homeassistant.core import HomeAssistant

from custom_components.peaqev.peaqservice.hub.hub import Hub
from custom_components.peaqev.peaqservice.util.constants import TYPELITE
from .const import (
    DOMAIN,
    PLATFORMS, LISTENER_FN_CLOSE
)
from .peaqservice.hub.hub_lite import HubLite

_LOGGER = logging.getLogger(__name__)

async def _get_existing_param(conf, parameter: str, default_val: any):
    if parameter in conf.options.keys():
        return conf.options.get(parameter)
    if parameter in conf.data.keys():
        return conf.data.get(parameter)
    return default_val

async def async_setup_entry(hass: HomeAssistant, conf: ConfigEntry) -> bool:
    """Set up Peaq"""

    hass.data.setdefault(DOMAIN, {})
    hass.data[DOMAIN][conf.entry_id] = conf.data

    if "peaqevtype" in conf.data.keys():
        peaqtype_is_lite = bool(conf.data["peaqevtype"] == TYPELITE)

    else:
        peaqtype_is_lite = False

    ci = {}

    ci["locale"] = conf.data["locale"]
    ci["chargertype"] = conf.data["chargertype"]
    ci["chargerid"] = conf.data["chargerid"]
    ci["startpeaks"] = conf.options["startpeaks"] if "startpeaks" in conf.options.keys() else conf.data["startpeaks"]
    ci["priceaware"] = await _get_existing_param(conf, "priceaware", False)
    ci["peaqtype_is_lite"] = peaqtype_is_lite

    if ci["priceaware"] is False:
        ci["cautionhours"] = conf.options["cautionhours"] if "cautionhours" in conf.options.keys() else conf.data["cautionhours"] if "cautionhours" in conf.data.keys() else []
        ci["nonhours"] = conf.options["nonhours"] if "nonhours" in conf.options.keys() else conf.data["nonhours"] if "nonhours" in conf.data.keys() else []
    else:
        ci["absolute_top_price"] = await _get_existing_param(conf, "absolute_top_price", 0)
        ci["min_price"] = await _get_existing_param(conf, "min_priceaware_threshold_price", 0)
        ci["cautionhour_type"] = conf.options["cautionhour_type"] if "cautionhour_type" in conf.options.keys() else conf.data["cautionhour_type"]

    if peaqtype_is_lite is True:
        hub = HubLite(hass, ci, DOMAIN)
    else:
        ci["powersensor"] = conf.data["name"]
        ci["powersensorincludescar"] = conf.data["powersensorincludescar"]

        hub = Hub(hass, ci, DOMAIN)

    hass.data[DOMAIN]["hub"] = hub

    async def servicehandler_enable(call): # pylint:disable=unused-argument
        await hub.call_enable_peaq()

    async def servicehandler_disable(call): # pylint:disable=unused-argument
        await hub.call_disable_peaq()

    hass.services.async_register(DOMAIN, "enable", servicehandler_enable)
    hass.services.async_register(DOMAIN, "disable", servicehandler_disable)

    hass.config_entries.async_setup_platforms(conf, PLATFORMS)

    undo_listener = conf.add_update_listener(config_entry_update_listener)

    hass.data[DOMAIN][conf.entry_id] = {
        LISTENER_FN_CLOSE: undo_listener,
    }
    return True

async def config_entry_update_listener(hass: HomeAssistant, conf: ConfigEntry):
    await hass.config_entries.async_reload(conf.entry_id)

async def async_unload_entry(hass: HomeAssistant, conf: ConfigEntry) -> bool:
    """Unload a config entry."""
    unload_ok = hass.config_entries.async_unload_platforms(conf, PLATFORMS)
    if unload_ok:
        hass.data[DOMAIN].pop(conf.entry_id)

    return unload_ok
```

andreasropel, 2022-06-07:
I also having this issue when trying to add my Easee wallbox with latest available version (0.10.1): 

```
Logger: custom_components.peaqev.peaqservice.chargertypes.chargerbase
Source: custom_components/peaqev/peaqservice/chargertypes/chargerbase.py:126 
Integration: peaqev ev-charging (documentation, issues) 
First occurred: 10:15:36 (1 occurrences) 
Last logged: 10:15:36

no entities!
```

elden1337, 2022-06-07:
this is really strange. and it doesn't rectify itself later either?
the domain it is looking for is "easee", same on as is set on the integration i'm trying to work against: 

https://github.com/fondberg/easee_hass/blob/e2694582ebac15aecc4edb4dc821226d48cb45ca/custom_components/easee/manifest.json#L2

is equal to this:
https://github.com/elden1337/hass-peaq/blob/32c2023ea178f9f990f3bddf38a47e0ab78280c7/custom_components/peaqev/peaqservice/chargertypes/types/easee.py#L31

I do get the same issue at first when re-installing (perhaps i'm too early in checking it), but it then suceeds to do all the things it's supposed and initialize the hub properly, see below flow:

```
2022-06-07 12:04:29 ERROR (MainThread) [custom_components.peaqev.peaqservice.chargertypes.chargerbase] no entities!
2022-06-07 12:04:29 INFO (MainThread) [custom_components.peaqev.peaqservice.chargertypes.chargerbase] {'chargerentity': 'sensor.hake_[SECRET]m_1', 'powermeter': 'sensor.hake_[SECRET]m_1_power', 'powermeter_factor': 1, 'powerswitch': 'switch.hake_[SECRET]m_1_2', 'ampmeter': 'max_current', 'ampmeter_is_attribute': True, 'servicecalls_on': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7f85159670>, 'servicecalls_off': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7f851595e0>, 'servicecalls_resume': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7f85159670>, 'servicecalls_pause': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7f851595e0>, 'updatecurrent': 'set_max_current'}
2022-06-07 12:04:29 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] Chargerobject-state not found in given state-list. Value was: None
2022-06-07 12:04:29 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubdata.hubdatabase] sensor.hake_[SECRET]m_1
2022-06-07 12:04:29 INFO (MainThread) [custom_components.peaqev.peaqservice.hourselection] Hourselection has initialized
2022-06-07 12:04:29 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] Carpowersensor has initialized
2022-06-07 12:04:29 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] switch.hake_[SECRET]m_1_2 has initialized
2022-06-07 12:04:29 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] None has initialized
2022-06-07 12:04:29 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] sensor.nibe_power has initialized
2022-06-07 12:04:29 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] Chargerobject has initialized
2022-06-07 12:04:29 INFO (MainThread) [custom_components.peaqev.peaqservice.chargecontroller.chargecontrollerbase] Chargecontroller is initialized and ready to work!
```

I'm having a hard time seeing what the issue is.

elden1337, 2022-06-07:
if you have access to the file-system, can you check in config/.storage/core.entity_registry if there are any items holding "platform": "easee" ? this is what is queried basically.

elden1337, 2022-06-07:
```
service: easee.set_charger_dynamic_limit
data:
  charger_id: AB123456
  current: 6
```

the same as would be used as charger_id in this service-call. I'm unsure what that would be called in easee-world.

andreasropel, 2022-06-07:
Then I am using correct. Shall I reinstall the Easee integration perhaps?

elden1337, 2022-06-07:
hmm, i really don't know. the call i'm doing is to a HA-core-type to get all entities bound to a domain. and if it can't find them it seems to halt in your case, but not in the chargeamps-case. I'll have to check for deviations between those two chargertypes first (as you saw above, mine also calls this, but then manages anyway, so the logging error is a bit weird on my behalf as well).

elden1337, 2022-06-07:
by all means, try if you want to. but i think there's a bug in my code that i just can't figure out.

andreasropel, 2022-06-07:
Reinstalled and it did not rectified the issue. :/

elden1337, 2022-06-07:
> Reinstalled and it did not rectified the issue. :/ Anyone else having this issue?

i bet all with easee at the moment unfortunately.

andreasropel, 2022-06-07:
Seems to work. I can see the entities. :)

elden1337, 2022-06-07:
great!

andreasropel, 2022-06-07:
Does this look ok? Q: Why is the 'sensor.peaqev_hour_controller' disabled? 

[image]

elden1337, 2022-06-07:
Looks better than i've seen from any easee-user before at least ;). 

hour_controller should only be created if one is using price-awareness. sure you picked that on re-install?

elden1337, 2022-06-07:
i don't love the minus values though... do you have solar?

andreasropel, 2022-06-07:
Yes, I've chosen to use price-awareness function in setup. That's strange it isn't enabled then. Nordpool integration is active.

Yes, I do have solar production. I am eagerly awaiting the additional stuff you've entered into your peaq-board. :D

elden1337, 2022-06-07:
logs don't say anything weird about prices or hours? like it hasn't been able to implement it properly?

andreasropel, 2022-06-07:
This is what the logs are saying:
```
2022-06-07 17:21:53 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubdata.hubdatabase] sensor.easeeidredacted_status
2022-06-07 17:21:54 INFO (MainThread) [custom_components.peaqev.peaqservice.hourselection] Hourselection has initialized
2022-06-07 17:21:54 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] Carpowersensor has initialized
2022-06-07 17:21:54 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] switch.eh7wacn6_is_enabled has initialized
2022-06-07 17:21:54 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] sensor.ferroamp_grid_power has initialized
2022-06-07 17:21:54 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] sensor.peaqev_house_power has initialized
2022-06-07 17:21:54 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] Chargerobject has initialized
2022-06-07 17:21:54 INFO (MainThread) [custom_components.peaqev.peaqservice.chargecontroller.chargecontrollerbase] Chargecontroller is initialized and ready to work!
2022-06-07 17:21:54 ERROR (MainThread) [custom_components.peaqev.peaqservice.hub.hubbase] Unable to handle data: binary_sensor.peaqev_charger_done (extra keys not allowed @ data['mode'])
```

elden1337, 2022-06-07:
That last log entry is a new bug at least. 

The hourcontroller. How does it behave if you reboot Ha?

reinestenberg, 2022-06-07:
I also managed to install it now.
I can see all entities as in andreasropels post.
After installation, when I chose "Configure" on Peaq in Settings->Integrations I get this message:
[image]

I choose default settings: non price awareness, Intermediate on first page. Nothing in drop-downs on second page

andreasropel, 2022-06-07:
> That last log entry is a new bug at least.
> 
> The hourcontroller. How does it behave if you reboot Ha?

After reboot: 

`2022-06-07 18:34:10 INFO (MainThread) [custom_components.peaqev.peaqservice.chargertypes.chargerbase] entityschema is: easeeidredacted at 1654619650.8072484
2022-06-07 18:34:10 INFO (MainThread) [custom_components.peaqev.peaqservice.chargertypes.chargerbase] {'chargerentity': 'sensor.easeeidredacted_status', 'powermeter': 'sensor.easeeidredacted_power', 'powermeter_factor': 1000, 'powerswitch': 'switch.easeeidredacted_is_enabled', 'ampmeter': 'sensor.easeeidredacted_max_charger_limit', 'ampmeter_is_attribute': False, 'servicecalls_on': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7fc4a0178280>, 'servicecalls_off': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7fc4a01782e0>, 'servicecalls_resume': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7fc4a0178280>, 'servicecalls_pause': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7fc4a01782e0>, 'updatecurrent': 'set_charger_dynamic_limit'}
2022-06-07 18:34:10 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] Chargerobject-state not found in given state-list. Value was: None
2022-06-07 18:34:10 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubdata.hubdatabase] sensor.easeeidredacted_status
2022-06-07 18:34:10 ERROR (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.chargerswitch] [unavailable] could not set value as chargercurrent. invalid literal for int() with base 10: 'unavailable'
2022-06-07 18:34:11 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] Carpowersensor has initialized
2022-06-07 18:34:11 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] switch.easeeidredacted_is_enabled has initialized
2022-06-07 18:34:11 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] sensor.ferroamp_grid_power has initialized
2022-06-07 18:34:11 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] sensor.peaqev_house_power has initialized
2022-06-07 18:34:11 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 18:34:11 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 18:34:11 WARNING (SyncWorker_6) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 18:34:12 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 18:34:12 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 18:34:12 WARNING (SyncWorker_6) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 18:34:12 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 18:34:12 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 18:34:12 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 18:34:12 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 18:34:12 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 18:34:12 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 18:34:12 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] Chargerobject has initialized
2022-06-07 18:34:12 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours'] has not initialized yet.
2022-06-07 18:34:12 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours'] has not initialized yet.
2022-06-07 18:34:12 INFO (MainThread) [custom_components.peaqev.peaqservice.hourselection] Hourselection has initialized
2022-06-07 18:34:12 ERROR (MainThread) [custom_components.peaqev.peaqservice.hub.hubbase] Unable to handle data: sensor.nordpool_kwh_se3_sek_2_00_0 (extra keys not allowed @ data['mode'])
2022-06-07 18:34:18 INFO (SyncWorker_1) [custom_components.peaqev.peaqservice.chargecontroller.chargecontrollerbase] Chargecontroller is initialized and ready to work!`

elden1337, 2022-06-07:
ok, something's up with latest HA-core version. i just installed it and now i can't get it to work either.

elden1337, 2022-06-07:
@andreasropel Which version of HA are you using? That "extra keys not allowed" is a peculiar one.

andreasropel, 2022-06-07:
@elden1337 
Home Assistant Core 2022.6.3
Home Assistant Supervisor 2022.05.3
Home Assistant OS 7.6

elden1337, 2022-06-07:
one very interesting bit is that @reinestenberg seems to share an issue with a friend of mine, that top-log "chargerobject has not initialized yet". 

you don't have that issue @andreasropel ?! since you saw all the sensors earlier at least.

I wonder if easee has different versions of hardware that causes this on some installations but not all?

andreasropel, 2022-06-07:
I've reinstalled both Nordpool and Peaq, got some other errors listed in log here: 

`2022-06-07 20:49:32 ERROR (MainThread) [custom_components.peaqev.peaqservice.hourselection] could not get nordpool-prices
2022-06-07 20:49:32 INFO (MainThread) [custom_components.peaqev.peaqservice.chargertypes.chargerbase] entityschema is: easeeidredacted at 1654627772.138259
2022-06-07 20:49:32 INFO (MainThread) [custom_components.peaqev.peaqservice.chargertypes.chargerbase] {'chargerentity': 'sensor.easeeidredacted_status', 'powermeter': 'sensor.easeeidredacted_power', 'powermeter_factor': 1000, 'powerswitch': 'switch.easeeidredacted_is_enabled', 'ampmeter': 'sensor.easeeidredacted_max_charger_limit', 'ampmeter_is_attribute': False, 'servicecalls_on': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7fa7e8612190>, 'servicecalls_off': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7fa7e86121f0>, 'servicecalls_resume': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7fa7e8612190>, 'servicecalls_pause': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7fa7e86121f0>, 'updatecurrent': 'set_charger_dynamic_limit'}
2022-06-07 20:49:32 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] Chargerobject-state not found in given state-list. Value was: None
2022-06-07 20:49:32 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubdata.hubdatabase] sensor.easeeidredacted_status
2022-06-07 20:49:32 ERROR (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.chargerswitch] [unavailable] could not set value as chargercurrent. invalid literal for int() with base 10: 'unavailable'
2022-06-07 20:49:32 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] Carpowersensor has initialized
2022-06-07 20:49:32 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] switch.easeeidredacted_is_enabled has initialized
2022-06-07 20:49:32 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] sensor.ferroamp_grid_power has initialized
2022-06-07 20:49:32 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] sensor.peaqev_house_power has initialized
2022-06-07 20:49:32 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 20:49:32 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 20:49:32 WARNING (SyncWorker_6) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 20:49:33 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 20:49:33 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 20:49:33 INFO (MainThread) [custom_components.peaqev.peaqservice.hourselection] Hourselection has initialized
2022-06-07 20:49:33 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['chargerobject'] has not initialized yet.
2022-06-07 20:49:33 WARNING (SyncWorker_0) [custom_components.peaqev.peaqservice.hub.hub] ['chargerobject'] has not initialized yet.
2022-06-07 20:49:33 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['chargerobject'] has not initialized yet.
2022-06-07 20:49:33 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['chargerobject'] has not initialized yet.
2022-06-07 20:49:33 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['chargerobject'] has not initialized yet.
2022-06-07 20:49:33 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['chargerobject'] has not initialized yet.
2022-06-07 20:49:33 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['chargerobject'] has not initialized yet.
2022-06-07 20:49:33 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['chargerobject'] has not initialized yet.
2022-06-07 20:49:33 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['chargerobject'] has not initialized yet.
2022-06-07 20:49:33 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] Chargerobject has initialized
2022-06-07 20:49:33 ERROR (MainThread) [custom_components.peaqev.peaqservice.hub.hubbase] Unable to handle data: sensor.easeeidredacted_status (extra keys not allowed @ data['mode'])
2022-06-07 20:49:39 INFO (SyncWorker_8) [custom_components.peaqev.peaqservice.chargecontroller.chargecontrollerbase] Chargecontroller is initialized and ready to work!
2022-06-07 20:50:40 ERROR (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.chargerswitch] [unavailable] could not set value as chargercurrent. invalid literal for int() with base 10: 'unavailable'`

When it comes to easee custom component I have installed version 0.9.33. Should I update?

elden1337, 2022-06-07:
yes please do, it has to work with the latest and i suspect something with their new statuses is making peaq go nuts (vacuuming the easee-code right now)

elden1337, 2022-06-07:
nordpool should be fine btw. perhaps i am unclear in the logging-work there. It usually comes alive after a while. that integration does not seem to finalize before it tells HA it's done, and therefor peaq continues with its setup (it's supposed to wait for nordpool)

elden1337, 2022-06-07:
glad you guys are helping out here though! @reinestenberg @andreasropel , way more fun to squash bugs when not being alone ;)

andreasropel, 2022-06-07:
Updated and one step closer.
Got the same error with `chargerobject`.

`2022-06-07 21:04:20 ERROR (MainThread) [custom_components.peaqev.peaqservice.hourselection] could not get nordpool-prices
2022-06-07 21:04:20 INFO (MainThread) [custom_components.peaqev.peaqservice.chargertypes.chargerbase] entityschema is:  at 1654628660.496703
2022-06-07 21:04:20 INFO (MainThread) [custom_components.peaqev.peaqservice.chargertypes.chargerbase] {'chargerentity': 'sensor._status', 'powermeter': 'sensor._power', 'powermeter_factor': 1000, 'powerswitch': 'switch._is_enabled', 'ampmeter': 'sensor._max_charger_limit', 'ampmeter_is_attribute': False, 'servicecalls_on': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7fa7d4ec4d30>, 'servicecalls_off': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7fa7d4ec4d90>, 'servicecalls_resume': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7fa7d4ec4d30>, 'servicecalls_pause': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7fa7d4ec4d90>, 'updatecurrent': 'set_charger_dynamic_limit'}
2022-06-07 21:04:20 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] Chargerobject-state not found in given state-list. Value was: None
2022-06-07 21:04:20 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubdata.hubdatabase] sensor._status
2022-06-07 21:04:20 ERROR (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.chargerswitch] chargerobject state was none
2022-06-07 21:04:21 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] Carpowersensor has initialized
2022-06-07 21:04:21 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] switch._is_enabled has initialized
2022-06-07 21:04:21 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] sensor.ferroamp_grid_power has initialized
2022-06-07 21:04:21 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] sensor.peaqev_house_power has initialized
2022-06-07 21:04:21 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 21:04:21 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 21:04:21 WARNING (SyncWorker_2) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 21:04:22 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 21:04:22 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 21:04:22 WARNING (SyncWorker_4) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 21:04:22 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 21:04:22 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 21:04:22 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 21:04:22 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 21:04:22 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 21:04:22 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject'] has not initialized yet.
2022-06-07 21:04:22 INFO (MainThread) [custom_components.peaqev.peaqservice.hourselection] Hourselection has initialized
2022-06-07 21:04:22 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['chargerobject'] has not initialized yet.
2022-06-07 21:04:23 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['chargerobject'] has not initialized yet.
2022-06-07 21:04:28 WARNING (SyncWorker_5) [custom_components.peaqev.peaqservice.hub.hub] ['chargerobject'] has not initialized yet.
2022-06-07 21:04:28 WARNING (SyncWorker_1) [custom_components.peaqev.peaqservice.hub.hub] ['chargerobject'] has not initialized yet.`

elden1337, 2022-06-07:
HA! this is new (for me at least, perhaps you also got that log @reinestenberg but missed it):

```
2022-06-07 21:04:20 INFO (MainThread) [custom_components.peaqev.peaqservice.chargertypes.chargerbase] entityschema is:  at 1654628660.496703
2022-06-07 21:04:20 INFO (MainThread) [custom_components.peaqev.peaqservice.chargertypes.chargerbase] {'chargerentity': 'sensor._status', 'powermeter': 'sensor._power', 'powermeter_factor': 1000, 'powerswitch': 'switch._is_enabled', 'ampmeter': 'sensor._max_charger_limit', 'ampmeter_is_attribute': False, 'servicecalls_on': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7fa7d4ec4d30>, 'servicecalls_off': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7fa7d4ec4d90>, 'servicecalls_resume': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7fa7d4ec4d30>, 'servicecalls_pause': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7fa7d4ec4d90>, 'updatecurrent': 'set_charger_dynamic_limit'}
2022-06-07 21:04:20 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] Chargerobject-state not found in given state-list. Value was: None
```

You clearly see here that entityschema is blank! or did you just clear it out @andreasropel ?

andreasropel, 2022-06-07:
Did not cleared it @elden1337

I removed the old easee custom component, 
removed the peaq integration,
restarted HA, 
installed newer easee custom component,
restarted HA,
set up easee integration,
set up peaq integration,
restarted HA.

elden1337, 2022-06-07:
any possibility anyone of you two have access to file-system?
if so, can you go to config/.storage/core.entity_registry, open it in text editor and search for one of your easee-sensors.

I need to know what it says there under the "platform"-key. It should say easee but i suspect it does not.

reinestenberg, 2022-06-07:
It says: easee

andreasropel, 2022-06-07:
Every key-value pair is saying `"platform": "easee"`

reinestenberg, 2022-06-07:
When configuring and entering the ID of the charger I used the short string EHRABRG
Not the device_id shown in the text from andreasropel above.
Not sure which one it should be

elden1337, 2022-06-07:
> When configuring and entering the ID of the charger I used the short string EHRABRG (because this is what is shown in the easee app) Not the device_id shown in the text from andreasropel above. Not sure which one it should be

should be the one you did i think. what i am unable to wrap my head around is why this worked with @andreasropel s previous version of easee, but not now. Since both apparently are setting stuff in the same place i'm looking 😭

andreasropel, 2022-06-07:
[image]

andreasropel, 2022-06-07:
With easee 0.9.41 I got Peaqev Hour controller 'unavailable', but logs says: 

```
2022-06-07 21:52:11 ERROR (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.chargerswitch] chargerobject state was none
2022-06-07 21:52:11 INFO (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] Chargerobject has initialized
```

elden1337, 2022-06-07:
@andreasropel now with it working, can you check the same config-entry again that you sent before. 
I'm particularly interested in the:

`"original_name": "EASEE Online", `
and 
`"unique_id": "EASEE_online",`

perhaps they are not capitalized?

andreasropel, 2022-06-07:
```
"original_name": "EASEE Online",
and
"unique_id": "EASEE_online"
```

They are capitalized

elden1337, 2022-06-07:
thanks. pushing an attempt to fix soon, gotta go to bed.

elden1337, 2022-06-07:
hmm, yeah this is suspicious! i would assume that the ha-core methods utilized should get by platform, but if they do get by the start of the unique_id i'm toast (or down for a new refinement). 

The fix pushed soon will include a retry, since i suspect there could possibly be that the integration calls ready before it is (like the nordpool) and the init of this type was only done once before.

andreasropel, 2022-06-07:
yes ok. 
FYI I can see every easee sensor includes the charger_id as prefix.

elden1337, 2022-06-07:
> yes ok. FYI I can see every easee sensor includes the charger_id as prefix.

that's fine. for chargeamps the sensors aren't called sensor.chargeamps_xxxx either. 
this is why i need this call, for a smoother installation process for users. 

the workaround is to force the input of charger_id or other hardcoded path to consume the sensors, but my aim has always been to lessen the manual input as much as i can.

andreasropel, 2022-06-07:
I do understand that!

elden1337, 2022-06-07:
https://github.com/elden1337/hass-peaq/releases/tag/v0.11.0

we'll see what happens now. otherwise i'll continue another night. 

good night.

elden1337, 2022-06-07:
already know it didn't fix this issue unfortunately. back to drawing board.

andreasropel, 2022-06-07:
it does work!
[image]

elden1337, 2022-06-07:
not really unfortunately. better user-experience than before, but "hub not ready. check logs!" means that one or many (in this case because of the pattern i'm quite sure) has not been initialized.

if you want to, you can replace the easee.py in custom_components/peaqev/peaqservice/chargertypes/types with this code. it's a hax, but given the comment above there, if the chargerid is not set to be the original name, and that is what HA is actually checking, it might work as fallback:

```
import logging

from homeassistant.core import HomeAssistant
from peaqevcore.Models import CHARGERSTATES

from custom_components.peaqev.peaqservice.chargertypes.calltype import CallType
from custom_components.peaqev.peaqservice.chargertypes.chargerbase import ChargerBase
from custom_components.peaqev.peaqservice.util.constants import (
    CHARGER,
    CHARGERID,
    CURRENT,
)

_LOGGER = logging.getLogger(__name__)

ENTITYENDINGS = [
    "_power",
    "_status",
    "_dimmer",
    "_downlight",
    "_lifetime_energy",
    "_online",
    "_current",
    "_voltage",
    "_output_limit",
    "_cost_per_kwh",
    "_enable_idle_current"
]

NATIVE_CHARGERSTATES = [
    "disconnected",
    "awaiting_start",
    "charging",
    "ready_to_charge",
    "completed",
    "error"
]

DOMAINNAME = "easee"
UPDATECURRENT = True
#docs: https://github.com/fondberg/easee_hass

class Easee(ChargerBase):
    def __init__(self, hass: HomeAssistant, chargerid, auth_required: bool = False):
        super().__init__(hass)
        self._chargerid = chargerid
        self._auth_required = auth_required
        self._domainname = DOMAINNAME
        self._entityendings = ENTITYENDINGS
        self._native_chargerstates = NATIVE_CHARGERSTATES
        self._chargerstates[CHARGERSTATES.Idle] = ["disconnected"]
        self._chargerstates[CHARGERSTATES.Connected] = ["awaiting_start", "ready_to_charge"]
        self._chargerstates[CHARGERSTATES.Charging] = ["charging"]
        self._chargerstates[CHARGERSTATES.Done] = ["completed"]
        self.getentities()
        if len(self._entityschema) < 1:
            self.getentities(domain=self._chargerid)
        self.set_sensors()

        servicecall_params = {
            CHARGER: "charger_id",
            CHARGERID: self._chargerid,
            CURRENT: "current"
        }

        _on_off_params = {"charger_id": self._chargerid}

        _on = CallType("start", _on_off_params)
        _off = CallType("stop", _on_off_params)
        _resume = CallType("set_charger_dynamic_limit", {"mode": "6", "charger_id": self._chargerid})
        _pause = CallType("set_charger_dynamic_limit", {"mode": "0", "charger_id": self._chargerid})

        self._set_servicecalls(
            domain=DOMAINNAME,
            on_call=_on if self._auth_required is True else _resume,
            off_call=_off if self._auth_required is True else _pause,
            pause_call=_pause,
            resume_call=_resume,
            allowupdatecurrent=UPDATECURRENT,
            update_current_call="set_charger_dynamic_limit",
            update_current_params=servicecall_params
        )

    def set_sensors(self):
        self.chargerentity = f"sensor.{self._entityschema}_status"
        self.powermeter = f"sensor.{self._entityschema}_power"
        self.powermeter_factor = 1000
        self.powerswitch = f"switch.{self._entityschema}_is_enabled"
        self.ampmeter = f"sensor.{self._entityschema}_max_charger_limit"
        self.ampmeter_is_attribute = False

```

andreasropel, 2022-06-07:
Thanks. Will try tomorrow. Nice work so far.

reinestenberg, 2022-06-08:
With latest version + code above I get this:
[image]

[image]

I don't know if these two lines in the log is from peaq:
2022-06-08 07:10:21 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547882601392] Received invalid command: energy/info
2022-06-08 07:10:21 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547882601392] Received invalid command: energy/info

[10 more replies]

closed 2022-06-09 by elden1337