elden1337/hass-peaq #82: Own Energy Production

stejo388, 2022-06-03:
Hi,
To make this integration even better my suggestion is to take into account own energy production.  For me that has solar panel it´s cheapest to charge the car when they produce energy independent on the hourly price at that moment.

To make it even more advanced you could look at the forecast to plan ahead.

elden1337, 2022-06-03:
Thank you. Have been thinking about this as well. 

There will probably have to be a setting for:

- Always favor charging on prod
- Favor charging if spotprice < x
- Delay charging if projected prod later today (as you mention). Etc.  

This will most def be a set of features, an epic. 

Do you code? If so would you like to help out?

This will touch both this repo and the core-repo i'm making for this.

stejo388, 2022-06-03:
Hi,

Sorry I have not really coded for 15 years so I don´t think you would like to have my help :) But I´m glad to help you trying and discussing the topic.

elden1337, 2022-06-03:
Cool. More people have started to gather around this repo now which is awesome, so perhaps someone else will take on this (faster delivery than just me coding). 

Ill start when i have time to split this into stories. 

I guess you have your production visible in HA then so peaq can see it?!

andreasropel, 2022-06-03:
Yes, you can see the forecast through the forecast.solar integration in HA. It creates sensor(s) for today and tomorrow.

elden1337, 2022-06-06:
- [ ]  #83
- [ ]   #84

closed 2024-10-26 by elden1337