elden1337/hass-peaq #106: Peaq not adjusting Easee charger limit

rabakaas, 2022-06-09:
Eventually got Peaq installed but seems like its not able to control Easee dynamic charger limit.
Peaq allowed current is adjusting, but not the charger.
Charger is responding to manual service call: easee.set_charger_dynamic_limit

[image]

elden1337, 2022-06-09:
Great, thank you for this! (As you may have seen ive been struggeling gravely with Easee since i cannot qa it here). 

I'll have a look!

elden1337, 2022-06-09:
Is peaq stopping and starting it as expected?

elden1337, 2022-06-09:
If you check the logs, do you see any logs from peaq saying that it was calling update-current?

elden1337, 2022-06-09:
Checked with other easee owners. 

Have you turned off any tibber-controls for easee (if you have tibber)? 

Apparently they are setting aswell if integration is on.

mejje, 2022-06-09:
Can you provide a screenshot of the Easee app "Energy restrictions"?
[image]

rabakaas, 2022-06-09:
No Tibber, Peaq starting and stopping as it should.

Peaqev Charger controller endret til Start
19:00:01 - For 2 timer siden
Peaqev Charger controller endret til Stop
18:29:44 - For 3 timer siden
Peaqev Charger enabled slått på
17:58:15 - For 3 timer siden

Logger: custom_components.peaqev.peaqservice.hub.hubmember.chargerswitch
Source: custom_components/peaqev/peaqservice/hub/hubmember/chargerswitch.py:52
Integration: peaqev ev-charging (documentation, issues)
First occurred: 16:14:12 (4 occurrences)
Last logged: 19:08:36

chargerobject state was none

Source: custom_components/peaqev/peaqservice/chargecontroller/chargecontrollerbase.py:42
Integration: peaqev ev-charging (documentation, issues)
First occurred: 19:08:34 (10 occurrences)
Last logged: 19:08:34

Chargecontroller returned faulty state. Charger reported unavailable as state.

I`m a bit new to HA so if its something specific I should check I  will probably need some guidance :-)

[image]

elden1337, 2022-06-09:
I'll dive in and see if i can figure it out with this info. 

_Also, on a sidenote, fantastic to see a norwegian friend entering the club! Välkommen!_

mejje, 2022-06-09:
No, this one, and then expand "Restrictions"
[image]

rabakaas, 2022-06-09:
Aaah.
[image]
.

mejje, 2022-06-09:
Perfect, thanks for that. Now we wait for the code warrior.

elden1337, 2022-06-09:
does this attribute in your installation match the currently reported current of the charger (ie not peaq but the easee-charger?) that would be the first step to make sure i can actually read it properly.

[image]

rabakaas, 2022-06-09:
No, it doesnt match. Reports 6A when charger is at 32.

[image]

elden1337, 2022-06-09:
Good (or bad, but good ;)), then i have first steps to attend to. Since it cannot read it, it won't be able to properly set it either.

elden1337, 2022-06-09:
what would be the entity name for this one? 

[image]

I currently have it set as: `sensor.{self._entityschema}_max_charger_limit` where entityschema is your id there. but that clearly must be the wrong sensor for this. is it perhaps `sensor.{self._entityschema}_dynamic_charger_limit` ?

rabakaas, 2022-06-09:
Yes, its sensor.easee_dynamic_charger_limit

elden1337, 2022-06-09:
> Yes, its sensor.easee_dynamic_charger_limit

did you manually enable that sensor? @mejje has this sensor disabled by the integration in his installation.

rabakaas, 2022-06-09:
Sorry, don`t remember if I enabled it manually, long time since I installed it, but I don`t think so.
These was disabled by the integration, but enabled manually by me earlier today to test...

EHWV5Z55 Max charger limit
sensor.easee_max_charger_limit
Easee EV Charger
Garasje
—
—

EHWV5Z55 Max circuit limit
sensor.easee_max_circuit_limit
Easee EV Charger
Garasje
—
—

EHWV5Z55 Offline circuit limit
sensor.easee_offline_circuit_limit
Easee EV Charger
Garasje
—
—

EHWV5Z55 Output limit
sensor.easee_output_limit
Easee EV Charger
Garasje
—
—

EHWV5Z55 Voltage
sensor.easee_voltage
Easee EV Charger
Garasje
—
—

EHWV5Z55 Weekly schedule
binary_sensor.easee_weekly_schedule
Easee EV Charger
Garasje
—
—

sensor.easee_cost_per_kwh
sensor.easee_cost_per_kwh
Easee EV Charger
Garasje
Integrasjon

sensor.easee_reason_for_no_current
sensor.easee_reason_for_no_current
Easee EV Charger
Garasje

rabakaas, 2022-06-09:
No idea why, but some of the sensors for easee integration is named like this:
sensor.**ehwv5z55**_status
and some like this:
sensor.**easee**_dynamic_charger_limit

mejje, 2022-06-09:
Unfortunately, https://github.com/fondberg/easee_hass/blob/master/custom_components/easee/const.py#L281 indicates it's not enabled by default.

elden1337, 2022-06-09:
Might have to do a workaround for this.

elden1337, 2022-06-10:
Have release a service-release with an attempted fix. let me know if it works better soon:
https://github.com/elden1337/hass-peaq/releases/tag/v0.12.1

rabakaas, 2022-06-10:
Thanks for your effort.
Deleted my Easee integration and reinstalled.
All my Easee entities are now named like this with the name of the charger, sensor.**ehwv5z55**_dynamic_circuit_limit
Updated Peaq and restarted.
Had to manually enable sensors in Easee integration to get Peaq to start.

Seems like Peaq now reads the Charger reported current, but if I manually adjust the dynamic_circuit_limit it does not update unless I reload the Easee integration. Then the Charger reported current updates to the new value. I asume the Charger reported current is supposed to update itself when the current is changing? 

Please let me know if there is anything I can do to help.

rabakaas, 2022-06-10:
Does this help? Car is fuly charged, but can try a session later in the evening..

2022-06-10 17:17:48 DEBUG (MainThread) [custom_components.peaqev.peaqservice.chargertypes.chargerbase] entities discovered for easee are: ['binary_sensor.ehwv5z55_cable_locked', 'binary_sensor.ehwv5z55_online', 'sensor.ehwv5z55_status', 'sensor.ehwv5z55_power', 'sensor.ehwv5z55_session_energy', 'sensor.ehwv5z55_lifetime_energy', 'sensor.ehwv5z55_energy_per_hour', 'sensor.ehwv5z55_output_limit', 'sensor.ehwv5z55_current', 'sensor.ehwv5z55_circuit_current', 'sensor.ehwv5z55_dynamic_circuit_limit', 'sensor.ehwv5z55_max_circuit_limit', 'sensor.ehwv5z55_dynamic_charger_limit', 'sensor.ehwv5z55_offline_circuit_limit', 'sensor.ehwv5z55_max_charger_limit', 'sensor.ehwv5z55_voltage', 'switch.ehwv5z55_smart_charging', 'switch.ehwv5z55_cable_locked_permanently', 'switch.ehwv5z55_is_enabled', 'switch.ehwv5z55_enable_idle_current']
2022-06-10 17:17:48 DEBUG (MainThread) [custom_components.peaqev.peaqservice.chargertypes.chargerbase] entityschema is: ehwv5z55 at 1654874268.0490785
2022-06-10 17:17:48 DEBUG (MainThread) [custom_components.peaqev.peaqservice.chargertypes.chargerbase] {'chargerentity': 'sensor.ehwv5z55_status', 'powermeter': 'sensor.ehwv5z55_power', 'powermeter_factor': 1000, 'powerswitch': 'switch.ehwv5z55_is_enabled', 'ampmeter': 'sensor.ehwv5z55_dynamic_charger_limit', 'ampmeter_is_attribute': False, 'servicecalls_on': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7f9a454610>, 'servicecalls_off': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7f9a454670>, 'servicecalls_resume': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7f9a454610>, 'servicecalls_pause': <custom_components.peaqev.peaqservice.chargertypes.calltype.CallType object at 0x7f9a454670>, 'updatecurrent': 'set_charger_dynamic_limit'}
2022-06-10 17:17:48 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] Chargerobject-state not found in given state-list. Value was: None
2022-06-10 17:17:48 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.hubdata.hubdatabase] sensor.ehwv5z55_status
2022-06-10 17:17:48 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.peaqev
2022-06-10 17:17:48 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.peaqev
2022-06-10 17:17:48 INFO (MainThread) [homeassistant.components.switch] Setting up switch.peaqev
2022-06-10 17:17:48 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] sensor.aidon_active_power_import has initialized
2022-06-10 17:17:48 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] sensor.peaqev_house_power has initialized
2022-06-10 17:17:48 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['carpowersensor', 'chargerobject_switch', 'chargerobject'] has not initialized yet.
2022-06-10 17:17:52 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] Chargerobject has initialized
2022-06-10 17:17:52 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] CarPower-sensor has initialized
2022-06-10 17:17:52 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.hubmember.hubmember] switch.ehwv5z55_is_enabled has initialized
2022-06-10 17:17:52 INFO (MainThread) [custom_components.peaqev.peaqservice.charger.charger] Calling charger off
2022-06-10 17:17:58 INFO (SyncWorker_1) [custom_components.peaqev.peaqservice.chargecontroller.chargecontrollerbase] Chargecontroller is initialized and ready to work!
022-06-10 17:18:05 INFO (MainThread) [homeassistant.components.automation.peaq_charge] Initialized trigger Peaq Charge
2022-06-10 17:18:05 INFO (MainThread) [homeassistant.components.automation.peaq_disconnected] Initialized trigger Peaq Disconnected
2022-06-10 17:18:54 INFO (MainThread) [custom_components.peaqev.peaqservice.charger.charger] Calling charger off
2022-06-10 17:18:54 ERROR (MainThread) [custom_components.easee.services] Could not find charger ehwv5z55
2022-06-10 17:18:54 ERROR (MainThread) [homeassistant.core] Error executing service: <ServiceCall easee.set_charger_dynamic_limit (c:01G5734SESJSFVSHV882W6F89C): current=0, charger_id=ehwv5z55>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1722, in catch_exceptions
    await coro_or_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1741, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/config/custom_components/easee/services.py", line 397, in charger_execute_set_current
    raise HomeAssistantError("Could not find charger {}".format(charger_id))
homeassistant.exceptions.HomeAssistantError: Could not find charger ehwv5z55
2022-06-10 17:19:09 INFO (MainThread) [custom_components.peaqev.peaqservice.charger.charger] charger-class has been stopped

elden1337, 2022-06-10:
Very good. And looks quite ok. That last log seems a bit weird. Is there a connection issue with easee perhaps? What do you think @mejje ?

mejje, 2022-06-12:
I've had a bunch of 502's from easee the last couple of days, so I wouldn't be surprised. Unless the charger id is case sensitive? mine is uppercase.

elden1337, 2022-06-13:
if possible, test the latest beta-version https://github.com/elden1337/hass-peaq/releases/tag/v0.13.0-beta.6 and see if that one works better for you @rabakaas

elden1337, 2022-06-15:
Thanks! 

Make sure to update to latest version first. I know have seen other being able to charge with Easee as well.

elden1337, 2022-06-19:
@rabakaas i've gotten reports from other easee users now of this working well. Closing this issue for now. Please re-open if you still experience issues.

closed 2022-06-19 by elden1337

rabakaas, 2022-06-19:
Finally got time to install latest version and test, seems to be working now.
Thanks!

elden1337, 2022-06-19:
At the moment, yes. 

Glad you got it working! Looking forward to reports or thoughts when you get ideas.