elden1337/hass-peaq #115: "Your car"-settings

elden1337, 2022-06-20:
Allow user to set up their car in peaqev. 

- capacity
- general departure time
- generally needed charge, this can help peaqev chose cheap enough hours through the night. 

Servicecall that can override these settings?

uphillbattle, 2022-10-06:
For BMW, the API rate limitation is pretty strict (the Home Assistant BMW integration recently had to reduce the number of API calls since users experience being locked out). Therefore, going a route in which Peaqev enquires the API itself may «overload» the API. Since many cars (for example BMW) have working integrations in Home Assistant, I recommend utilizing the already existing sensors in HA (by entering sensor name/ID information at Peaqev setup).

elden1337, 2022-10-07:
Hi, absolutely, the intention is to utilize existing integrations for this, like i'm doing with chargertypes and nordpool