elden1337/hass-peaq #151: Enable three-phase charging.

trfollegg, 2022-10-06:
When using Peaqev ev-charging with Easee it will not enable three-phase charging.

andreasropel, 2022-10-06:
I think it does. At least on my site. Can you please ensure your Easee is set to use three phases?

trfollegg, 2022-10-06:
It was set to automatic. It works with three-phase when it is locked to it.
If you have to lock it to three-phase it becomes problematic when you have two cars, and one only takes one phase.

elden1337, 2022-10-07:
Hmm, it should actually adapt to what the charger reports and set itself to 1 or 3 phase depending on the incoming wattage. 

What happens for you? You connect and it stays put at around 3.5kw?

trfollegg, 2022-10-07:
Yes, it stays at around 3.5kw on 1 phase if it is set to automatic on the charger.

elden1337, 2022-10-07:
Hmm. 
Is it easee? If so all peaqev does on start is changing amps to 7A (since that's necessary to wake it up), then it probably runs an update amps to 6A. 

If chargeamps it calls the service to enable it. 

I don't have an easee myself but dont do anything else. 

The only possible thing i could think of is my treshold for determining 1 or 3, i might have to alter that, but that wouldnt change your charger, just the way peaqev looks at how many amps it can allow for a certain period in time.

elden1337, 2022-10-07:
I might have to debug somehow. But am away til sunday.

trfollegg, 2022-10-07:
It is Easee. Now it only works with 1 phase. I can see that it gives the 7A to start the charger. And then it goes back to 6A.
If I set the scheduler at let's say charge_amount: 6.3 departure_time: 22-10-07 12:50, it starts and sets the current to 16A.
After a short time, it goes back to 6A.
Maybe it is the car that refuses to start if the charger is locked to 3 phases and not automatic.

elden1337, 2022-10-08:
Yes i dont see how this can occur from peaq. Please double check and charge without peaq. Does it go 3-phase then?

trfollegg, 2022-10-08:
If I start the charger with service easee.resume it starts charging on 3-phase. It works like using the Easee app.

elden1337, 2022-10-11:
I asked another user with easee to check. Here are images from his webcloud when started through peaqev: 

[image]

[image]

trfollegg, 2022-10-11:
It looks like it changes depending on the amount it can use. I used intermediate and 6.3 peak now and then it used 3 phase when the charging amount was higher. you can see in the history.
[image]

elden1337, 2022-10-11:
I have an idea that i can test, but still find it very weird that easee would go one phase based on the amp-numbers i ask for.

elden1337, 2022-10-11:
new release is out now. I'll assign you this issue, please close it if it works better for you now, otherwise give me a shout so that I know it's still wonky.

andreasropel, 2022-10-12:
@trfollegg Can you please verify you have a three phase cable? Is it possible to charge on three phases without peaq enabled?

trfollegg, 2022-10-12:
@andreasropel Yes I have a 3-phase cable.
It looks like the update helped. It now charges on 3-phases.
[image]

closed 2022-10-12 by trfollegg