elden1337/hass-peaq #152: switch.peaqev_charger_enabled turns off when HA is restarted

uphillbattle, 2022-10-07:
`switch.peaqev_charger_enabled` seems to turn off after restarting Home Assistant.

The logs show these warnings:
```
2022-10-07 07:45:46.897 WARNING (MainThread) [peaqevcore.models.hub.chargerobject] Chargerobject-state not found in given state-list. Value was: None
2022-10-07 07:45:46.900 WARNING (MainThread) [peaqevcore.models.hub.chargerswitch] Chargerobject state was None while getting current. Updatecurrent will still process but you may see a missmatch in frontend.
2022-10-07 07:45:46.991 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['hours', 'chargerobject_switch', 'chargerobject'] has not initialized yet.
2022-10-07 07:45:47.168 WARNING (MainThread) [custom_components.peaqev.peaqservice.hub.hub] ['chargerobject_switch', 'chargerobject'] has not initialized yet.
```

elden1337, 2022-10-07:
Have you set the option to enable peaqev after reboot? By default it should not be enabled. 

If you have and this problem persists ill have a look, thus setting this as bug for now.

uphillbattle, 2022-10-07:
Ah, I had overlook that option. Enabled the option now. I'll close the issue and re-report it if issue persists (which I'm sure it won't). 

Thanks. 😄

closed 2022-10-07 by uphillbattle