elden1337/hass-peaq #154: Unable to load hass-peaq

anoteng, 2022-10-08:
I have added the integration through HACS, restarted HA and can find the integration in the normal HA integration add, but get an error trying to load it:
`This integration does not support configuration via the UI. If you followed this link from the Home Assistant website, make sure you run the latest version of Home Assistant.`

Here are the relevant parts of the log:
```
2022-10-08 18:28:05.584 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_components.peaqev.config_flow
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.9/site-packages/homeassistant/loader.py", line 681, in get_platform
    cache[full_name] = self._import_platform(platform_name)
  File "/srv/homeassistant/lib/python3.9/site-packages/homeassistant/loader.py", line 698, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/homeassistant/.homeassistant/custom_components/peaqev/__init__.py", line 10, in <module>
    from custom_components.peaqev.peaqservice.hub.hub import HomeAssistantHub
  File "/home/homeassistant/.homeassistant/custom_components/peaqev/peaqservice/hub/hub.py", line 10, in <module>
    from peaqevcore.hub.hub import Hub
  File "/srv/homeassistant/lib/python3.9/site-packages/peaqevcore/hub/hub.py", line 1, in <module>
    from ..services.hourselection.hourselectionfactory import HourselectionFactory
  File "/srv/homeassistant/lib/python3.9/site-packages/peaqevcore/services/hourselection/hourselectionfactory.py", line 1, in <module>
    from .price_aware_hours import PriceAwareHours
  File "/srv/homeassistant/lib/python3.9/site-packages/peaqevcore/services/hourselection/price_aware_hours.py", line 3, in <module>
    from .hoursselection import Hoursselection as core_hours
  File "/srv/homeassistant/lib/python3.9/site-packages/peaqevcore/services/hourselection/hoursselection.py", line 153
    match listtype:
          ^
SyntaxError: invalid syntax
2022-10-08 18:28:05.591 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration peaqev: Exception importing custom_components.peaqev.config_flow
2022-10-08 18:30:21.713 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_components.peaqev.config_flow
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.9/site-packages/homeassistant/loader.py", line 681, in get_platform
    cache[full_name] = self._import_platform(platform_name)
  File "/srv/homeassistant/lib/python3.9/site-packages/homeassistant/loader.py", line 698, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 972, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/homeassistant/.homeassistant/custom_components/peaqev/__init__.py", line 10, in <module>
    from custom_components.peaqev.peaqservice.hub.hub import HomeAssistantHub
  File "/home/homeassistant/.homeassistant/custom_components/peaqev/peaqservice/hub/hub.py", line 10, in <module>
    from peaqevcore.hub.hub import Hub
  File "/srv/homeassistant/lib/python3.9/site-packages/peaqevcore/hub/hub.py", line 1, in <module>
    from ..services.hourselection.hourselectionfactory import HourselectionFactory
  File "/srv/homeassistant/lib/python3.9/site-packages/peaqevcore/services/hourselection/hourselectionfactory.py", line 1, in <module>
    from .price_aware_hours import PriceAwareHours
  File "/srv/homeassistant/lib/python3.9/site-packages/peaqevcore/services/hourselection/price_aware_hours.py", line 3, in <module>
    from .hoursselection import Hoursselection as core_hours
  File "/srv/homeassistant/lib/python3.9/site-packages/peaqevcore/services/hourselection/hoursselection.py", line 153
    match listtype:
          ^
SyntaxError: invalid syntax
2022-10-08 18:30:21.727 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration peaqev: Exception importing custom_components.peaqev.config_flow
```

elden1337, 2022-10-09:
Hi, It looks like you need to update Home assistant, the errors you are getting tells me that you are still running python 3.9, and some code in peaqev is too modern for that. 

I think they changed to python 3.10 in like the july-release or something. 

Please update and get back to me if working/not working.

anoteng, 2022-10-09:
Ok, that worked. Had to migrate my HA into docker, as I wasn't ready to upgrade my debian machine also searving as a router to testing.

closed 2022-10-09 by anoteng