elden1337/hass-peaq #168: Help with setup

kirilldem, 2022-11-03:
Hello,

Thanks for the addon, it sounds really useful!

During the setup, I have added my glow sensor that measures power usage in real time along with my status of the Easee charger (_sensor.carport_status_ - the one that says charging, ready_to_charge) as _Charger id_.

When I enable the on the peaq charge nothing happens. Am I missing something?

No nordpool pricing is selected or limited hours of operation.

elden1337, 2022-11-04:
Hey, glad you found us! ;). 

It seems as if you ve added a faulty charger id. It will not be the sensor from HA but the s/n value written on your easee when it was new and now in your guide. I think about 8 characters long. 

Since this item is not an optional and the setup has failed you will need to uninstall - install peaqev. 

Best regards, Magnus

kirilldem, 2022-11-04:
> Hey, glad you found us! ;).
>
> It seems as if you ve added a faulty charger id. It will not be the sensor from HA but the s/n value written on your easee when it was new and now in your guide. I think [...]

Thanks for the feedback and that makes sense :)

I removed the integration, restarted it and then went through the setup again. I added my serial number to Charge id. Then I turn on the peaqev charge, but the car still draws ~3.5kw. How long should it take to reduce the charge speed?
Am I missing something?

elden1337, 2022-11-04:
1, when using easee people often have other snart charging active (ie Tibber). This must be turned off otherwise that will try to start it all the time. 

2. What does the sensors say? Does chargecontroller say "Start"? If so the charger is allowed to charge. 
What does the allowed current state?