elden1337/hass-peaq #173: Make sure non price aware works in > 2.0

elden1337, 2022-11-05:
- Session cost sensor must be ignored
- Check init of hub in core for nordpool issues
- More?

closed 2022-11-05 by elden1337

kirilldem, 2022-11-05:
I now get all the entities, will give it some time for it to average my usage :)

kirilldem, 2022-11-05:
still missing some entitities, so it does not work:

[image]

I tried to add to existing integration the Nordpool pricing option but it errored:
[image]

Next i delete the integration, reboot and add again but with nordpool pricing, I get more sensors but for some reason it thinks i shall not charge untill 2000 (I set 100 as threshold and no hours blocked) here are all the sensors I see:

[image]

I try to update it to 1000 from configure button (integration menu) but same error as before.

This is what the integration sees with charging running at 3.7kw and house pulling 4.5kw

[image]

elden1337, 2022-11-05:
okay, getting closer @kirilldem. A couple of things for the getting-started wiki that I can see here:

1. Max and Min-prices are not to determine that the car can or cannot charge. The algorithm is calculating that itself based on nordpool's fluctuations. Those parameters are merely an extra gate like "i will never charge if price is above 5, and will always let it charge if it's below 0.2". So the "charging stopped until 20:00" is still correct.
2. The eeror you get on options i cannot say why. Did the logs give you any hint?
3. Now when it's been running for a while i guess the sensors make more sense? One thing that I can see is that you have checked that your powersensor does include the car (i see it on your sensors being what they are). I hope this is correct cos otherwise it will calculate badly.

The max-price can be set at 0 if you don't care about it _(will add in getting-started)_

To be honest, if you don't have any funky logs i think this should be working for you now? Which area do you live in, so that i know what kind of peak-charge I calculate for you.

these are my sensors, running powersensor excluding car, with price awareness:

[image]

elden1337, 2022-11-05:
[image]

Here you can see that 4.5 - 3.8 = ca 700-800w which is what the sensor is stating. So i guess you have made an error by checking the "powersensor includes car". I think your powersensor is set after your carcharger and that the car is not included?

kirilldem, 2022-11-06:
> okay, getting closer @kirilldem. A couple of things for the getting-started wiki that I can see here:
>
> 1. Max and Min-prices are not to determine that the car can or cannot charge. The algorithm is [...]

Still does not work unfortunately :(

This is the error I get when changing nordpool prove on the "configure" button in the integration page once peaqev is already setup. 

```
Logger: aiohttp.server
Source: custom_components/peaqev/config_flow.py:243
Integration: peaqev ev-charging ([documentation](https://github.com/elden1337/hass-peaq), [issues](https://github.com/elden1337/hass-peaq/issues))
First occurred: 8:09:24 PM (1 occurrences)
Last logged: 8:09:24 PM

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 82, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 136, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 241, in post
    return await super().post(request, flow_id)
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 73, in wrapper
    result = await method(view, request, data, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 110, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 280, in async_configure
    result = await self._async_handle_step(
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 367, in _async_handle_step
    result: FlowResult = await getattr(flow, method)(user_input)
  File "/config/custom_components/peaqev/config_flow.py", line 182, in async_step_init
    return await self.async_step_months()
  File "/config/custom_components/peaqev/config_flow.py", line 243, in async_step_months
    vol.Optional("jan", default=defaultvalues["1"]): cv.positive_float,
KeyError: '1'
```

My power sensor measures the usage at the meter by seeing how many blinks the meter is making and how often they are happening. So in that sense the power sensor will include the car power usage too. 

I took a video of what is happening, so you can better see how it works. 
I turn on the peaqev charging on the 16th second, you can't quite see it so well in the video. 

https://user-images.githubusercontent.com/44521162/200190281-7a522cf5-e3e4-40fa-80db-f75b2c51e353.mp4

kirilldem, 2022-11-06:
I just tried one more time with peaqev turned on and the car stops charging after about 30 seconds. 

No errors in the log.

elden1337, 2022-11-06:
Very ambitious, thank you!

The log error has nothing to do with nordpool. you seem to have some error in the months (namely January given the log). 

As for the video: 

1. in the beta you are running i have discovered one flaw, that causes peaqev to stop the charger even if it is disabled. That will no longer be the case in beta2.
2. Other than that, It seems to behave as it is supposed to. You see, the charging is supposed to be stopped until 21:00, therefore, if you turn peaqev on it will stop the charger. This is by design and the reason why I can give you cheaper hours (same would go if the peak is about to be breached, then the charger is stopped as well).

I will soon deploy beta2, but overall i think this looks like it's working quite well for you.

elden1337, 2022-11-06:
Also, i've come to realize that Easee does start itself on connecting the car. So the feature of overtaking from peaqev and stopping (or commencing but taking control) is paramount to it working with Easee. I have Chargeamps myself and that one is disabled in between the sessions.

kirilldem, 2022-11-06:
> Also, i've come to realize that Easee does start itself on connecting the car. So the feature of overtaking from peaqev and stopping (or commencing but taking control) is paramount to it working with Easee. I have Chargeamps myself and that one is disabled in between the sessions.

I had a look in the settings and it is not possible to delay the charging unless I set charging hours in my car. There are some smart charging options included in the easee app, but i have not checked those out yet.

elden1337, 2022-11-06:
> I had a look in the settings and it is not possible to delay the charging unless I set charging hours in my car. There are some smart charging options included in the easee app, but i have not [...]

No what I mean is that you should not have to. As long as any smart-charging capabilities and schedules on easee are turned off (or tibber or other vendor controlling easee if the user has that) it should be all good.

kirilldem, 2022-11-06:
Alright, then I wait for the next beta :)

elden1337, 2022-11-06:
It's ready: https://github.com/elden1337/hass-peaq/releases/tag/v2.2.0-beta.2

kirilldem, 2022-11-06:
We are getting there :) Have a look at the video, it is not throttling at all now, but maybe that's by design.

[Overview-–-Home-Assistant (1).webm](https://user-images.githubusercontent.com/44521162/200194342-a2e802de-8983-4969-9f0d-7c9df34d2e47.webm)

elden1337, 2022-11-06:
Cant see the video (perhaps since im on mobile) but yes it should be able to throttle, depending on your peak and current predicitons. 

This is mine since i connected it with beta2. 

[image]

elden1337, 2022-11-07:
Ok how did it go? 
The beta worked fine here with chargeamps, and have reports from easee-users of it working for them as well.

kirilldem, 2022-11-07:
The car is now fully charged, will try later on. What is the sensor name that calculates average hourly peak?

elden1337, 2022-11-07:
sensor.peaqev_peak

elden1337, 2022-11-08:
So @kirilldem is the beta a go from your end or you found issues I need to know about?

kirilldem, 2022-11-08:
I am not sure actually, still get charging breaks from time to time. I am continuing to test. Did her this yesterday though 👍
[image]

elden1337, 2022-11-08:
Charging breaks as it stops for a while and then starts again? 
This is by design. How else could i help stay below the peak?

kirilldem, 2022-11-08:
It stops and then never comes back. But I need to test more

elden1337, 2022-11-08:
Can u show me the history of the relevant sensors?

- sensor.peaqev_hour_controller
- sensor.peaqev_charger_controller

kirilldem, 2022-11-09:
Tried it again earlier today and it still breaks the charging shortly after turning it on:

https://user-images.githubusercontent.com/44521162/200929813-81196f89-2414-427b-ae78-1284843c9e5a.mp4

https://user-images.githubusercontent.com/44521162/200929821-f6e9535e-b5c0-49fa-8fdc-ee4b048502e1.mp4

elden1337, 2022-11-09:
Movie 1 and 2 seem perfectly correct. 
The controller says "stop", and also states that it's stopped til 21. So what happens when it detects the charger charging is that it is telling it to pause. 

Screenshot of hours. 
The hours are made from the nordpool-pricing automatically. Since it's after 21 now the next stop is at 7am according to your notes. 

Why it is done on the last screenshot i can not say, i guess it charged some now after 21? 

This is the way peaq should stop and start charging. Im going to release next version soon.

kirilldem, 2022-11-09:
> Movie 1 and 2 seem perfectly correct. The controller says "stop", and also states that it's stopped til 21. So what happens when it detects the charger charging is that it is telling it to pause. [...]

That makes sense but my confusion comes because as per your earlier advice I set norpool pricing to 0, since i dont have an hourly electricity agreement. 

I think there was 3 percent or so left to charge so it probably did do some top up in the evening.

elden1337, 2022-11-09:
you mean max-price? that is not disabling nordpool, that is just telling that you don't want a specific static top-price over which no charging should ever be allowed.

I'll adjust the getting-started. Thank you so much for testing!

kirilldem, 2022-11-09:
I've updated it like that, hopefully it will stop preventing it from charging :)

[image]

elden1337, 2022-11-09:
yes, make an issue for it if they somehow dont.
and update to regular 2.2.0 now, i'll start working on next release.