elden1337/hass-peaq #176: get_average_kwh_price_core could not be calculated: division by zero

rawrgit, 2022-11-09:
Found this warning flooding my logs every 5 seconds. Known issue?

This error originated from a custom integration.

Logger: peaqevcore.services.hourselection.hoursselection
Source: custom_components/peaqev/sensors/money_sensor.py:41
Integration: peaqev ev-charging (documentation, issues)
First occurred: 07:03:40 (5542 occurrences)
Last logged: 13:13:11

get_average_kwh_price_core could not be calculated: division by zero

elden1337, 2022-11-09:
Not known afaik. Which version are you running? And do you have price-awareness active?

rawrgit, 2022-11-09:
Did update to the latest (the one you put out about an hour ago) and it seems like those logs stopped flooding.

Yes, I do have price-awareness active.

elden1337, 2022-11-09:
Ok cool. Then I hope it shall not return

closed 2022-11-09 by elden1337