elden1337/hass-peaq #184: trying to set up a HomeWizard energy sensor into peaq but not getting further then this ...

dsmagghe, 2022-12-29:
2022-12-29 22:43:43.556 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry sensor.kwh_meter_3_phase_3c39e7253780_active_power for peaqev
File "/config/custom_components/peaqev/__init__.py", line 58, in async_setup_entry
File "/config/custom_components/peaqev/peaqservice/hub/hub.py", line 43, in __init__
File "/config/custom_components/peaqev/peaqservice/chargertypes/chargertypes.py", line 26, in __init__
File "/config/custom_components/peaqev/config_flow.py", line 227, in async_step_hours
File "/config/custom_components/peaqev/config_flow.py", line 262, in async_step_months
File "/config/custom_components/peaqev/config_flow.py", line 227, in async_step_hours
File "/config/custom_components/peaqev/config_flow.py", line 262, in async_step_months
File "/config/custom_components/peaqev/config_flow.py", line 227, in async_step_hours
File "/config/custom_components/peaqev/config_flow.py", line 262, in async_step_months
File "/config/custom_components/peaqev/config_flow.py", line 227, in async_step_hours
File "/config/custom_components/peaqev/config_flow.py", line 262, in async_step_months
File "/config/custom_components/peaqev/config_flow.py", line 227, in async_step_hours
File "/config/custom_components/peaqev/config_flow.py", line 262, in async_step_months
File "/config/custom_components/peaqev/config_flow.py", line 227, in async_step_hours
File "/config/custom_components/peaqev/config_flow.py", line 262, in async_step_months
File "/config/custom_components/peaqev/config_flow.py", line 227, in async_step_hours
File "/config/custom_components/peaqev/config_flow.py", line 262, in async_step_months
File "/config/custom_components/peaqev/config_flow.py", line 227, in async_step_hours
File "/config/custom_components/peaqev/config_flow.py", line 262, in async_step_months
2022-12-29 22:47:34.743 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry sensor.kwh_meter_3_phase_3c39e7253780_active_power for peaqev
File "/config/custom_components/peaqev/__init__.py", line 58, in async_setup_entry
File "/config/custom_components/peaqev/peaqservice/hub/hub.py", line 43, in __init__
File "/config/custom_components/peaqev/peaqservice/chargertypes/chargertypes.py", line 26, in __init__

elden1337, 2022-12-30:
Oups that looks bad. Is this all it's logging for you? Is it om the step where you enter the powersensor and click next?

elden1337, 2022-12-30:
found it. my bad. pushing hotfix

elden1337, 2022-12-30:
but fyi, i saw that you merged the pr for zaptec update amps. that won't be available just yet, need to do some changes to adhere to that once zaptec's been updated with those service calls.

elden1337, 2022-12-30:
https://github.com/elden1337/hass-peaq/releases/tag/v2.5.1

commit 67f2cc83 (2022-12-30): bump 2.5.1 and fix #184 (hopefully)
closed 2022-12-30 by elden1337