elden1337/hass-peaq #198: Installation without chargertype

elden1337, 2023-02-11:
To be able to utilize the sensors without an impmented chargertype. 
Would require user to write their own automations.

nord-, 2023-02-13:
Since I have a chargebox with a proprietary protocol I have to write the communication with it myself, and with access to the smart sensors the use-case for peaqev broadens to other things being able to control dishwasher, heating, swimming pools etc...

nord-, 2023-02-23:
Configure the integration fails with error "Unknown error occurred"
[image]

```
Felet härrör från en anpassad integration.

Logger: aiohttp.server
Source: custom_components/peaqev/config_flow.py:256
Integration: peaqev ev-charging (documentation, issues)
First occurred: 22 februari 2023 kl. 22:17:47 (6 occurrences)
Last logged: 10:48:50

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 80, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 145, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 241, in post
    return await super().post(request, flow_id)
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 72, in wrapper
    result = await method(view, request, data, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 110, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 253, in async_configure
    result = await self._async_handle_step(
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 335, in _async_handle_step
    result: FlowResult = await getattr(flow, method)(user_input)
  File "/config/custom_components/peaqev/config_flow.py", line 201, in async_step_init
    return await self.async_step_months()
  File "/config/custom_components/peaqev/config_flow.py", line 256, in async_step_months
    defaultvalues = {float(k): v for (k, v) in _defaultvalues}
  File "/config/custom_components/peaqev/config_flow.py", line 256, in <dictcomp>
    defaultvalues = {float(k): v for (k, v) in _defaultvalues}
TypeError: cannot unpack non-iterable int object
```

nord-, 2023-02-23:
The integration logs quite a few errors (1476 so far) 🙂.
```
Felet härrör från en anpassad integration.

Logger: custom_components.peaqev.peaqservice.hub.hub
Source: custom_components/peaqev/peaqservice/hub/hub.py:125
Integration: peaqev ev-charging (documentation, issues)
First occurred: 22 februari 2023 kl. 22:26:34 (1476 occurrences)
Last logged: 10:49:18

Unable to handle data: sensor.peaqev_energy_including_car_hourly (type object 'NoPeak' has no attribute 'query_model') <state sensor.peaqev_energy_including_car_hourly=0.66; state_class=total, source=sensor.peaqev_energy_including_car, status=collecting, last_period=0.90, meter_period=hourly, cron pattern=0 * * * *, last_reset=2023-02-23T09:00:00.009739+00:00, unit_of_measurement=kWh, device_class=energy, icon=mdi:counter, friendly_name=Peaqev energy_including_car hourly @ 2023-02-23T10:46:18.326359+01:00>|<state sensor.peaqev_energy_including_car_hourly=0.67; state_class=total, source=sensor.peaqev_energy_including_car, status=collecting, last_period=0.90, meter_period=hourly, cron pattern=0 * * * *, last_reset=2023-02-23T09:00:00.009739+00:00, unit_of_measurement=kWh, device_class=energy, icon=mdi:counter, friendly_name=Peaqev energy_including_car hourly @ 2023-02-23T10:47:18.439354+01:00>
Unable to handle data: sensor.peaqev_energy_including_car_hourly ('HubSensorsLite' object has no attribute 'carpowersensor') <state sensor.peaqev_energy_including_car_hourly=0.67; state_class=total, source=sensor.peaqev_energy_including_car, status=collecting, last_period=0.90, meter_period=hourly, cron pattern=0 * * * *, last_reset=2023-02-23T09:00:00.009739+00:00, unit_of_measurement=kWh, device_class=energy, icon=mdi:counter, friendly_name=Peaqev energy_including_car hourly @ 2023-02-23T10:47:18.439354+01:00>|<state sensor.peaqev_energy_including_car_hourly=0.68; state_class=total, source=sensor.peaqev_energy_including_car, status=collecting, last_period=0.90, meter_period=hourly, cron pattern=0 * * * *, last_reset=2023-02-23T09:00:00.009739+00:00, unit_of_measurement=kWh, device_class=energy, icon=mdi:counter, friendly_name=Peaqev energy_including_car hourly @ 2023-02-23T10:48:18.327606+01:00>
Unable to handle data: sensor.peaqev_energy_including_car_hourly (type object 'NoPeak' has no attribute 'query_model') <state sensor.peaqev_energy_including_car_hourly=0.67; state_class=total, source=sensor.peaqev_energy_including_car, status=collecting, last_period=0.90, meter_period=hourly, cron pattern=0 * * * *, last_reset=2023-02-23T09:00:00.009739+00:00, unit_of_measurement=kWh, device_class=energy, icon=mdi:counter, friendly_name=Peaqev energy_including_car hourly @ 2023-02-23T10:47:18.439354+01:00>|<state sensor.peaqev_energy_including_car_hourly=0.68; state_class=total, source=sensor.peaqev_energy_including_car, status=collecting, last_period=0.90, meter_period=hourly, cron pattern=0 * * * *, last_reset=2023-02-23T09:00:00.009739+00:00, unit_of_measurement=kWh, device_class=energy, icon=mdi:counter, friendly_name=Peaqev energy_including_car hourly @ 2023-02-23T10:48:18.327606+01:00>
Unable to handle data: sensor.peaqev_energy_including_car_hourly ('HubSensorsLite' object has no attribute 'carpowersensor') <state sensor.peaqev_energy_including_car_hourly=0.68; state_class=total, source=sensor.peaqev_energy_including_car, status=collecting, last_period=0.90, meter_period=hourly, cron pattern=0 * * * *, last_reset=2023-02-23T09:00:00.009739+00:00, unit_of_measurement=kWh, device_class=energy, icon=mdi:counter, friendly_name=Peaqev energy_including_car hourly @ 2023-02-23T10:48:18.327606+01:00>|<state sensor.peaqev_energy_including_car_hourly=0.69; state_class=total, source=sensor.peaqev_energy_including_car, status=collecting, last_period=0.90, meter_period=hourly, cron pattern=0 * * * *, last_reset=2023-02-23T09:00:00.009739+00:00, unit_of_measurement=kWh, device_class=energy, icon=mdi:counter, friendly_name=Peaqev energy_including_car hourly @ 2023-02-23T10:49:18.329722+01:00>
Unable to handle data: sensor.peaqev_energy_including_car_hourly (type object 'NoPeak' has no attribute 'query_model') <state sensor.peaqev_energy_including_car_hourly=0.68; state_class=total, source=sensor.peaqev_energy_including_car, status=collecting, last_period=0.90, meter_period=hourly, cron pattern=0 * * * *, last_reset=2023-02-23T09:00:00.009739+00:00, unit_of_measurement=kWh, device_class=energy, icon=mdi:counter, friendly_name=Peaqev energy_including_car hourly @ 2023-02-23T10:48:18.327606+01:00>|<state sensor.peaqev_energy_including_car_hourly=0.69; state_class=total, source=sensor.peaqev_energy_including_car, status=collecting, last_period=0.90, meter_period=hourly, cron pattern=0 * * * *, last_reset=2023-02-23T09:00:00.009739+00:00, unit_of_measurement=kWh, device_class=energy, icon=mdi:counter, friendly_name=Peaqev energy_including_car hourly @ 2023-02-23T10:49:18.329722+01:00>
```

elden1337, 2023-02-23:
Great! Working on it. Thanks for reporting.

elden1337, 2023-02-23:
> Configure the integration (after it been added) fails with error "Unknown error occurred"
> 
> [image]
> 
> 
> 
> ```
> 
> Felet härrör från en anpassad integration.
> 
> 
> 
> Logger: aiohttp.server
> 
> Source: custom_components/peaqev/config_flow.py:256
> 
> Integration: peaqev ev-charging (documentation, issues)
> 
> First occurred: 22 februari 2023 kl. 22:17:47 (6 occurrences)
> 
> Last logged: 10:48:50
> 
> 
> 
> Error handling request
> 
> Traceback (most recent call last):
> 
>   File "/usr/local/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
> 
>     resp = await request_handler(request)
> 
>   File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
> 
>     resp = await handler(request)
> 
>   File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
> 
>     return await handler(request)
> 
>   File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
> 
>     return await handler(request)
> 
>   File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
> 
>     return await handler(request)
> 
>   File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
> 
>     return await handler(request)
> 
>   File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 80, in ban_middleware
> 
>     return await handler(request)
> 
>   File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
> 
>     return await handler(request)
> 
>   File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 145, in handle
> 
>     result = await result
> 
>   File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 241, in post
> 
>     return await super().post(request, flow_id)
> 
>   File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 72, in wrapper
> 
>     result = await method(view, request, data, *args, **kwargs)
> 
>   File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 110, in post
> 
>     result = await self._flow_mgr.async_configure(flow_id, data)
> 
>   File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 253, in async_configure
> 
>     result = await self._async_handle_step(
> 
>   File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 335, in _async_handle_step
> 
>     result: FlowResult = await getattr(flow, method)(user_input)
> 
>   File "/config/custom_components/peaqev/config_flow.py", line 201, in async_step_init
> 
>     return await self.async_step_months()
> 
>   File "/config/custom_components/peaqev/config_flow.py", line 256, in async_step_months
> 
>     defaultvalues = {float(k): v for (k, v) in _defaultvalues}
> 
>   File "/config/custom_components/peaqev/config_flow.py", line 256, in <dictcomp>
> 
>     defaultvalues = {float(k): v for (k, v) in _defaultvalues}
> 
> TypeError: cannot unpack non-iterable int object
> 
> ```
> 
> 

This one's handled after beta was released.

elden1337, 2023-02-23:
> The integration logs quite a few errors (1476 so far) 🙂.
>
> ```
>
> Felet härrör från en anpassad integration.
>
>
>
> Logger: custom_components.peaqev.peaqservice.hub.hub
>
> Source: [...]

Hmmm, looks like you are setting up peaqev-lite (without power meter)? That wont work without any charger, or rather, the sensors would not do anything useful i'm afraid. Do you have a power meter connected?

nord-, 2023-02-24:
> Hmmm, looks like you are setting up peaqev-lite (without power meter)? That wont work without any charger, or rather, the sensors would not do anything useful i'm afraid. Do you have a power meter connected?

I started with setting up the lite-integration, but changed my mind halfway, aborted the installation and started over with a "full" setup. Perhaps something stayed on? Should I uninstall the integration and reboot HA and start over?

elden1337, 2023-02-24:
> I started with setting up the lite-integration, but changed my mind halfway, aborted the installation and started over with a "full" setup. Perhaps something stayed on? Should I uninstall the [...]

Yes please uninstall, reboot and reinstall (still have to fix those issues though but so that we are on a clean slate for your prefered setup). 

Also i saw u are doing nopeak, there is no peak charge where you live yet?

nord-, 2023-02-24:
> Or rather, I think the only thing that will work then is the hourselection. But that might be enough?

The hour-selection is enough, power monitor is done by the chargebox itself already (and I use HA for that for other appliances like heatpump and hottub), and I don't have peak power charge (yet).

nord-, 2023-02-24:
> Also i saw u are doing nopeak, there is no peak charge where you live yet?

No, no peak charge (Trollhättan Energi) yet.

nord-, 2023-02-24:
Tried it again, full installation this time, but left _chargerid_ blank (since I don't want PeaqEv to control it). Then I got this error after installation of integration:
```
Felet härrör från en anpassad integration.

Logger: homeassistant.config_entries
Source: custom_components/peaqev/__init__.py:40
Integration: peaqev ev-charging (documentation, issues)
First occurred: 15:36:43 (1 occurrences)
Last logged: 15:36:43

Error setting up entry sensor.template_house_power_w for peaqev
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 382, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/peaqev/__init__.py", line 40, in async_setup_entry
    options.charger.chargerid = conf.data["chargerid"]
KeyError: 'chargerid'
```

elden1337, 2023-02-24:
More entagled than id like it to be. This is what comes out of one year of concept coding. Thanks for reporting!

elden1337, 2023-02-28:
Have u had time to install beta2 @nord- ? Any thoughts?

closed 2023-02-28 by elden1337