elden1337/hass-peaq #205: Charger enabled

reinestenberg, 2023-03-10:
Yesterday I had to "panic charge" my car, since the public transport stopped working due to some snow.
This was effectively hindered by peaqev doing its job, so I disabled peaqev.

Charger type: Easee

peaqev charger controller == Stop
Set Peaqev_charger_enabled = False
Result: Charger dynamic current == 0
Desired result: Charger dynamic current == 16

Set Peaqev_charger_enabled = True
* Wait for "current percentage" to go below "start" * 
peaqev charger controller == Start
Set Peaqev_charger_enabled = False
Result: Charger dynamic current == 16
Desired result: Charger dynamic current == 16

elden1337, 2023-03-10:
If i understand correctly peaq continued to stop your charger even after you disabled and started charging? If so this is a bug. 

What happens when you disable peaq is that it calls the charger to turn off (to not cause undesired charging that goes unnoticed). However, you should then be able to start your charger and peaqev shouldnt care about it until enabled again. 

So what did happen in the end? Did it continue to stop easee?

reinestenberg, 2023-03-10:
It set the charger dynamic current to 0.
I was not able to start charging. The Easee app told me it was current limited and it was the charger dynamic current that was set to 0. 
Coming to think of it I might have mixed things up a bit. I used home assistant (developer tools) to set the charger dynamic current to 16 while on the tram so the family at home could charge the car. After doing this it did not stop from charging (although it did stop me from charging once before in a similar situation in december 2022)

elden1337, 2023-03-10:
Ill try to perform a similar test and see what happens. 

Desired is as above, if setting to disabled - turn off charging (once), then ignore the charger until enabled again.

reinestenberg, 2023-03-10:
How is it turned off? By setting current to 0 or by calling api (stop or pause or whatever)?

elden1337, 2023-03-10:
Peaqev has a switch. Peaqev_charger_enabled

Thats the only way to turn it on/off

elden1337, 2023-03-10:
A command is sent to the charger.py to turn it off. How that's done depends on the model. In easee it means that if there is no car it would call this:

https://github.com/elden1337/hass-peaq/blob/c7d055d3322434642d206257a604f0a97174eaf5/custom_components/peaqev/peaqservice/chargertypes/types/easee.py#L108

Otherwise this:

https://github.com/elden1337/hass-peaq/blob/c7d055d3322434642d206257a604f0a97174eaf5/custom_components/peaqev/peaqservice/chargertypes/types/easee.py#L122

elden1337, 2023-03-10:
As far as i've read that is what the native apis do as well. If i do terminate easee on disable there could be auth issues instead.

reinestenberg, 2023-03-10:
"pause" does not remove auth according to easee api doc.
https://developer.easee.cloud/reference/post_api-chargers-id-commands-pause-charging

As you say it limits the dynamic current to 0, but it might be a difference to use "pause" and to externally set the dynamic current to 0. 
I have seen the charger app saying something like: "Charger controlled by external system" or something similar

elden1337, 2023-03-10:
Ok. Not sure still. If it would be on pause and be resumed from its own app it would probably just set current to 7. Ill have to ask other easee users aswell, afraid ill break the cyclic peaqcharge if i change this without more tests.

reinestenberg, 2023-03-10:
If I remember correctly I can set the dynamic current limit in HA and it works. It is AFAIK no way to start it from the official app.

I can test when I get home if I can start it from HA with a call to "resume" from developer tools when peaqev has set dynamic current limit to 0. I tried that yesterday from a easee integration in HA, but that did not work.

More investigation is needed

reinestenberg, 2023-03-10:
By "works" I mean it starts chargning.

elden1337, 2023-03-10:
Assigning you meanwhile. Get back when you've had a chance to check.

elden1337, 2023-03-14:
Any luck here? @reinestenberg

elden1337, 2023-04-11:
closing this and asking you to re-open if you find the time to test with the latest versions @reinestenberg

closed 2023-04-11 by elden1337