elden1337/hass-peaq #223: Error while loading

Neled77, 2023-03-19:
Hi,

updated to your latest version to be able to set the peak amps. But i get this while using the setup.

Thanks

Neled

```
2023-03-19 14:34:32.371 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_compo          
nents.peaqev.config_flow                                                                                                        
Traceback (most recent call last):                                                                                              
  File "/usr/lib/python3.10/site-packages/homeassistant/loader.py", line 779, in get_platform                                   
    cache[full_name] = self._import_platform(platform_name)                                                                     
  File "/usr/lib/python3.10/site-packages/homeassistant/loader.py", line 796, in _import_platform                               
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")                                                          
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module                                                  
    return _bootstrap._gcd_import(name[level:], package, level)                                                                 
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import                                                               
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load                                                            
  File "<frozen importlib._bootstrap>", line 992, in _find_and_load_unlocked                                                    
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed                                                  
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import                                                               
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load                                                            
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked                                                   
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked                                                             
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module                                                       
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed                                                  
  File "/config/custom_components/peaqev/__init__.py", line 10, in <module>                                                     
    from custom_components.peaqev.peaqservice.hub.hub import HomeAssistantHub                                                   
  File "/config/custom_components/peaqev/peaqservice/hub/hub.py", line 15, in <module>                                          
    from custom_components.peaqev.peaqservice.chargertypes.chargertype_factory import ChargerTypeFactory                        
  File "/config/custom_components/peaqev/peaqservice/chargertypes/chargertype_factory.py", line 9, in <module>                  
    from custom_components.peaqev.peaqservice.chargertypes.types.easee import Easee                                             
  File "/config/custom_components/peaqev/peaqservice/chargertypes/types/easee.py", line 62                                      
    def max_amps(self): -> int:                                                                                                 
                        ^^                                                                                                      
SyntaxError: invalid syntax                                                                                                     
2023-03-19 14:34:32.402 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow fo          
r integration peaqev: Exception importing custom_components.peaqev.config_flow 
```

Neled77, 2023-03-19:
worked deleting everything

closed 2023-03-19 by Neled77

elden1337, 2023-03-21:
Hi @Neled77, looks like you cloned the main here. I suggest you install through HACS to get the proper checked versions.