elden1337/hass-peaq #224: AvG Price per KWH

Neled77, 2023-03-20:
Hi,

Thanks for your great addon!
seems your average price per KW needs to be corrected .

`Non hours:
  - '17'
  - '18'
  - '19'
  - '20'
  - '21'
  - '22'
  - 6⁺¹
  - 7⁺¹
  - 8⁺¹
  - 9⁺¹
Caution hours:
  '23': 33%
  0⁺¹: 31%
  5⁺¹: 33%
Current hour charge permittance: 100%
Avg price per kWh: 9.28 EUR
Max charge amount: 383.0 kWh
Nordpool average 7 days: 0 EUR
nordpool_average_this_month: 0 EUR
Nordpool average data:
  - 8.192
  - 14.358
  - 10.769
  - 10.508
  - 11.039
  - 11.204
offsets:
  today:
    '0': -0.05
    '1': -0.41
    '2': -0.48
    '3': -0.6
    '4': -0.59
    '5': -0.35
    '6': 0.42
    '7': 0.92
    '8': 0.93
    '9': 0.49
    '10': -0.1
    '11': -0.37
    '12': -0.82
    '13': -1
    '14': -0.91
    '15': -0.78
    '16': -0.39
    '17': 0.42
    '18': 1.19
    '19': 1.81
    '20': 1.25
    '21': 0.53
    '22': 0.19
    '23': -0.26
  tomorrow:
    '0': -0.1
    '1': -0.46
    '2': -0.49
    '3': -0.62
    '4': -0.56
    '5': -0.21
    '6': 0.29
    '7': 0.96
    '8': 0.87
    '9': 0.23
    '10': -0.35
    '11': -0.7
    '12': -0.9
    '13': -1
    '14': -0.78
    '15': -0.79
    '16': -0.66
    '17': -0.14
    '18': 0.7
    '19': 1.18
    '20': 1.31
    '21': -0.2
    '22': -0.58
    '23': -1
icon: mdi:car-clock
friendly_name: Peaqev Hour controller`

[image]

All the best

Neled

elden1337, 2023-03-20:
Hi, you are most welcome!

Guess you mean this attribute on the sensor "Avg price per kWh: 9.28 EUR" which yes, seems bad?

The later one you posted is just a helper to calc and store the gain/loss (i will hide that sensor when i have time). So ignore the raw rate on it.

elden1337, 2023-03-20:
@Neled77 can you post your pricelist for today and tomorrow here? i need them to make a test-case.

elden1337, 2023-03-20:
ah, i see the issue @Neled77 , you obv have your prices in cents... have to make localization of the prices.

elden1337, 2023-03-20:
also fixed so that you can see average for the week and month before a full week or full month has been ingested into peaqev (you see the avg-data there). 

Is that total-chargeamount really true? what peak have you set?

closed 2023-03-21 by elden1337

Neled77, 2023-03-21:
Yes, I sould have said that I downloaded your beta version to test it :)

Testing your changes and giving you feedback asap 👍

elden1337, 2023-03-21:
Ah yes. The session sensor aswell... always something extra ;). 

But the attributes you showed me the other day look correct?

Neled77, 2023-03-21:
Also another question. 
Active power sensor show negative value when using the grid and positive power when upping to the grid. 
I created a Active power inverted sensor to feed PeaqEV. Which one would PeaqEV suit best?
[image]

elden1337, 2023-03-21:
Inverted in your case.

elden1337, 2023-03-21:
I assume you will create tickets regarding solar prioritization etc. ;).

Neled77, 2023-03-21:
> I assume you will create tickets regarding solar prioritization etc. ;).

correct :)actually trying to set this up. But a have a good buffer (30KW batteries)

Neled77, 2023-03-21:
Also trying to fix this one first :)

2023-03-21 10:07:43.892 WARNING (SyncWorker_4) [custom_components.peaqev.peaqservice.chargertypes.types.easee] Unable to get max
 amps. The sensor sensor.ehqnelbk_max_charger_limit returned state None. Setting max amps to 16 til I get a proper state.

elden1337, 2023-03-21:
Does it resolve later? Or stuck there? It should only log this once if everything is fine