elden1337/hass-peaq #226: Fault filling up log, Unable to handle data-update

rabakaas, 2023-03-23:
Having trouble with Peaqev filling log, see below.
Tried a fresh install without luck. V2.8.0

Logger: custom_components.peaqev.peaqservice.hub.hub
Source: custom_components/peaqev/peaqservice/hub/hub.py:152
Integration: peaqev ev-charging (documentation, issues)
First occurred: 12:00:15 (3330 occurrences)
Last logged: 14:09:14

Unable to handle data-update: sensor.peaqev_charger_controller <state sensor.peaqev_charger_controller=Idle; price aware=False, non_hours=[], caution_hours=[], current_hour state=Charging permitted, scheduler_active=False, latest_internal_chargerstart=2023-03-23 14:08:54, icon=mdi:electric-switch, friendly_name=Peaqev Charger controller @ 2023-03-23T12:18:40.613836+01:00>|<state sensor.peaqev_charger_controller=Idle; price aware=False, non_hours=[], caution_hours=[], current_hour state=Charging permitted, scheduler_active=False, latest_internal_chargerstart=2023-03-23 14:08:58, icon=mdi:electric-switch, friendly_name=Peaqev Charger controller @ 2023-03-23T12:18:40.613836+01:00>. Exception: 'NordPoolUpdater' object has no attribute 'nordpool_entity'
Unable to handle data-update: sensor.peaqev_charger_controller <state sensor.peaqev_charger_controller=Idle; price aware=False, non_hours=[], caution_hours=[], current_hour state=Charging permitted, scheduler_active=False, latest_internal_chargerstart=2023-03-23 14:08:58, icon=mdi:electric-switch, friendly_name=Peaqev Charger controller @ 2023-03-23T12:18:40.613836+01:00>|<state sensor.peaqev_charger_controller=Idle; price aware=False, non_hours=[], caution_hours=[], current_hour state=Charging permitted, scheduler_active=False, latest_internal_chargerstart=2023-03-23 14:09:02, icon=mdi:electric-switch, friendly_name=Peaqev Charger controller @ 2023-03-23T12:18:40.613836+01:00>. Exception: 'NordPoolUpdater' object has no attribute 'nordpool_entity'
Unable to handle data-update: sensor.peaqev_charger_controller <state sensor.peaqev_charger_controller=Idle; price aware=False, non_hours=[], caution_hours=[], current_hour state=Charging permitted, scheduler_active=False, latest_internal_chargerstart=2023-03-23 14:09:02, icon=mdi:electric-switch, friendly_name=Peaqev Charger controller @ 2023-03-23T12:18:40.613836+01:00>|<state sensor.peaqev_charger_controller=Idle; price aware=False, non_hours=[], caution_hours=[], current_hour state=Charging permitted, scheduler_active=False, latest_internal_chargerstart=2023-03-23 14:09:06, icon=mdi:electric-switch, friendly_name=Peaqev Charger controller @ 2023-03-23T12:18:40.613836+01:00>. Exception: 'NordPoolUpdater' object has no attribute 'nordpool_entity'
Unable to handle data-update: sensor.peaqev_charger_controller <state sensor.peaqev_charger_controller=Idle; price aware=False, non_hours=[], caution_hours=[], current_hour state=Charging permitted, scheduler_active=False, latest_internal_chargerstart=2023-03-23 14:09:06, icon=mdi:electric-switch, friendly_name=Peaqev Charger controller @ 2023-03-23T12:18:40.613836+01:00>|<state sensor.peaqev_charger_controller=Idle; price aware=False, non_hours=[], caution_hours=[], current_hour state=Charging permitted, scheduler_active=False, latest_internal_chargerstart=2023-03-23 14:09:10, icon=mdi:electric-switch, friendly_name=Peaqev Charger controller @ 2023-03-23T12:18:40.613836+01:00>. Exception: 'NordPoolUpdater' object has no attribute 'nordpool_entity'
Unable to handle data-update: sensor.peaqev_charger_controller <state sensor.peaqev_charger_controller=Idle; price aware=False, non_hours=[], caution_hours=[], current_hour state=Charging permitted, scheduler_active=False, latest_internal_chargerstart=2023-03-23 14:09:10, icon=mdi:electric-switch, friendly_name=Peaqev Charger controller @ 2023-03-23T12:18:40.613836+01:00>|<state sensor.peaqev_charger_controller=Idle; price aware=False, non_hours=[], caution_hours=[], current_hour state=Charging permitted, scheduler_active=False, latest_internal_chargerstart=2023-03-23 14:09:14, icon=mdi:electric-switch, friendly_name=Peaqev Charger controller @ 2023-03-23T12:18:40.613836+01:00>. Exception: 'NordPoolUpdater' object has no attribute 'nordpool_entity'

elden1337, 2023-03-23:
Shit. Thanks for reporting!

elden1337, 2023-03-23:
if you have file editor access and want to test right now you can replace the custom_components/peaqev/peaqservice/hub/nordpool/nordpool.py with this file.
Otherwise i'll see if i can push a new beta tonight or this weekend.

```python
import asyncio
import logging
from datetime import datetime
from statistics import mean

import homeassistant.helpers.template as template

from custom_components.peaqev.peaqservice.hub.nordpool.nordpool_dto import NordpoolDTO
from custom_components.peaqev.peaqservice.hub.nordpool.nordpool_model import NordPoolModel

_LOGGER = logging.getLogger(__name__)

NORDPOOL = "nordpool"
AVERAGE_MAX_LEN = 31

class NordPoolUpdater:
    def __init__(self, hub, is_active: bool = True):
        self.model = NordPoolModel()
        self.hub = hub
        self._nordpool_entity:str = None
        if is_active:
            self._setup_nordpool()

    @property
    def currency(self) -> str:
        return self.model.currency

    @property
    def state(self) -> float:
        return self.model.state

    @state.setter
    def state(self, val) -> None:
        if self.model.state != val:
            self.model.state = val

    @property
    def average_month(self) -> float:
        return self.model.average_month

    @property
    def average_weekly(self) -> float:
        return self.model.average_weekly

    @property
    def average_data(self) -> list:
        return self.model.average_data

    @property
    def average_ready(self) -> bool:
        return len(self.model.average_data) >= AVERAGE_MAX_LEN

    async def update_nordpool(self):
        if self._nordpool_entity is not None:
            ret = self.hub.state_machine.states.get(self._nordpool_entity)
            _result = NordpoolDTO()
            if ret is not None:
                await _result.set_model(ret)
                if await self._update_set_prices(_result):
                    await self.hub.observer.async_broadcast("prices changed",[self.model.prices, self.model.prices_tomorrow])
            elif self.hub.is_initialized:
                _LOGGER.error("Could not get nordpool-prices")

    async def _update_average_month(self) -> None:
        _new = await self._get_average_async(datetime.now().day)
        if len(self.model.average_data) > 0 and self.model.average_month != _new:
[... 56 lines cut ...]
        if isinstance(new_val, float):
            rounded = round(new_val, 3)
            if len(self.model.average_data) == 0:
                self.model.average_data.append(rounded)
            elif self.model.average_data[-1] != rounded:
                self.model.average_data.append(rounded)
            await self._cap_average_data_length()

    async def _cap_average_data_length(self):
        while len(self.model.average_data) > AVERAGE_MAX_LEN:
            del self.model.average_data[0]

    async def _get_average_async(self, days: int) -> float:
        try:
            if len(self.model.average_data) > days:
                ret = self.model.average_data[-days:]
            elif len(self.model.average_data) == 0:
                return 0.0
            else:
                ret = self.model.average_data
            return round(mean(ret), 2)
        except Exception as e:
            _LOGGER.debug(f"Could not calculate average. indata: {self.model.average_data}, error: {e}")

```

elden1337, 2023-03-23:
Ok you are Not using it. Then i need to check some more. You are hitting functions you shouldnt reach with your settings.

elden1337, 2023-03-24:
i ran 2.8.1 through the night here without issues. have released a hotfix @rabakaas . Hopefully it is working better for you as well: https://github.com/elden1337/hass-peaq/releases/tag/v2.8.1

Otherwise please re-open this issue and comment.

closed 2023-03-24 by elden1337

elden1337, 2023-04-11:
Yeah i get it. This is why i need more betatesters im afraid :(. I understand the issue so will try and fix shortly