elden1337/hass-peaq #249: PeaqEV have not been working for me lately and no time to investigate.

elden1337, 2023-04-11:
PeaqEV have not been working for me lately and no time to investigate.
Tried new version today and still having issues, see log:

Logger: custom_components.peaqev.peaqservice.hub.hub
Source: custom_components/peaqev/peaqservice/hub/hub.py:156
Integration: peaqev ev-charging (documentation, issues)
First occurred: 16:33:33 (498 occurrences)
Last logged: 16:45:02

Unable to handle data-update: sensor.aidon_active_power_import <state sensor.aidon_active_power_import=3443; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash, friendly_name=Active power import (Q1+Q4) @ 2023-04-11T16:44:50.985621+02:00>|<state sensor.aidon_active_power_import=2959; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash, friendly_name=Active power import (Q1+Q4) @ 2023-04-11T16:44:52.443338+02:00>. Exception: 'RegularHours' object has no attribute '_scheduler'
Unable to handle data-update: sensor.aidon_active_power_import <state sensor.aidon_active_power_import=2959; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash, friendly_name=Active power import (Q1+Q4) @ 2023-04-11T16:44:52.443338+02:00>|<state sensor.aidon_active_power_import=2963; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash, friendly_name=Active power import (Q1+Q4) @ 2023-04-11T16:44:54.943210+02:00>. Exception: 'RegularHours' object has no attribute '_scheduler'
Unable to handle data-update: sensor.aidon_active_power_import <state sensor.aidon_active_power_import=2963; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash, friendly_name=Active power import (Q1+Q4) @ 2023-04-11T16:44:54.943210+02:00>|<state sensor.aidon_active_power_import=2967; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash, friendly_name=Active power import (Q1+Q4) @ 2023-04-11T16:44:57.443176+02:00>. Exception: 'RegularHours' object has no attribute '_scheduler'
Unable to handle data-update: sensor.aidon_active_power_import <state sensor.aidon_active_power_import=2967; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash, friendly_name=Active power import (Q1+Q4) @ 2023-04-11T16:44:57.443176+02:00>|<state sensor.aidon_active_power_import=3005; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash, friendly_name=Active power import (Q1+Q4) @ 2023-04-11T16:45:00.983071+02:00>. Exception: 'RegularHours' object has no attribute '_scheduler'
Unable to handle data-update: sensor.aidon_active_power_import <state sensor.aidon_active_power_import=3005; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash, friendly_name=Active power import (Q1+Q4) @ 2023-04-11T16:45:00.983071+02:00>|<state sensor.aidon_active_power_import=2991; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash, friendly_name=Active power import (Q1+Q4) @ 2023-04-11T16:45:02.444229+02:00>. Exception: 'RegularHours' object has no attribute '_scheduler'

_Originally posted by @rabakaas in https://github.com/elden1337/hass-peaq/issues/226#issuecomment-1503517185_

elden1337, 2023-04-12:
@rabakaas https://github.com/elden1337/hass-peaq/releases/tag/v2.9.2

kirilldem, 2023-04-12:
Updated to 2.9.2 and still Peaqev Charger controller Unknown. Logs:

```
Logger: homeassistant.helpers.entity
Source: custom_components/peaqev/peaqservice/hub/hub.py:242
Integration: peaqev ev-charging ([documentation](https://github.com/elden1337/hass-peaq), [issues](https://github.com/elden1337/hass-peaq/issues))
First occurred: 9:35:44 AM (241 occurrences)
Last logged: 9:51:44 AM

Update for sensor.peaqev_charger_controller fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 550, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 755, in async_device_update
    raise exc
  File "/config/custom_components/peaqev/sensors/peaq_sensor.py", line 47, in async_update
    ret = await self.hub.async_request_sensor_data(
  File "/config/custom_components/peaqev/peaqservice/hub/hub.py", line 242, in async_request_sensor_data
    "is_scheduler_active": getattr(self.hours.scheduler, "scheduler_active"),
  File "/usr/local/lib/python3.10/site-packages/peaqevcore/services/hourselection/initializers/hoursbase.py", line 41, in scheduler
    return self._scheduler
AttributeError: 'RegularHours' object has no attribute '_scheduler'

```

```
Logger: custom_components.peaqev.peaqservice.hub.hub
Source: custom_components/peaqev/peaqservice/hub/hub.py:150
Integration: peaqev ev-charging ([documentation](https://github.com/elden1337/hass-peaq), [issues](https://github.com/elden1337/hass-peaq/issues))
First occurred: 9:35:11 AM (1846 occurrences)
Last logged: 9:51:48 AM

Unable to handle data-update: sensor.house_power_consumption <state sensor.house_power_consumption=687; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T09:51:45.410171+02:00>|<state sensor.house_power_consumption=686; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T09:51:45.944743+02:00>. Exception: 'RegularHours' object has no attribute '_scheduler'
Unable to handle data-update: sensor.house_power_consumption <state sensor.house_power_consumption=686; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T09:51:45.944743+02:00>|<state sensor.house_power_consumption=690; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T09:51:46.460919+02:00>. Exception: 'RegularHours' object has no attribute '_scheduler'
Unable to handle data-update: sensor.house_power_consumption <state sensor.house_power_consumption=690; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T09:51:46.460919+02:00>|<state sensor.house_power_consumption=688; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T09:51:46.976249+02:00>. Exception: 'RegularHours' object has no attribute '_scheduler'
Unable to handle data-update: sensor.house_power_consumption <state sensor.house_power_consumption=688; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T09:51:46.976249+02:00>|<state sensor.house_power_consumption=692; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T09:51:47.498861+02:00>. Exception: 'RegularHours' object has no attribute '_scheduler'
Unable to handle data-update: sensor.house_power_consumption <state sensor.house_power_consumption=692; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T09:51:47.498861+02:00>|<state sensor.house_power_consumption=690; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T09:51:48.021717+02:00>. Exception: 'RegularHours' object has no attribute '_scheduler'
```

kirilldem, 2023-04-12:
After amending the line to:

`            "is_scheduler_active": getattr(self.hours.scheduler, "scheduler_active", False),`
I still get the following errors and no change to the controller status

```
Logger: homeassistant.helpers.entity
Source: custom_components/peaqev/peaqservice/hub/hub.py:242
Integration: peaqev ev-charging ([documentation](https://github.com/elden1337/hass-peaq), [issues](https://github.com/elden1337/hass-peaq/issues))
First occurred: 10:10:15 AM (176 occurrences)
Last logged: 10:21:56 AM

Update for sensor.peaqev_charger_controller fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 550, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 755, in async_device_update
    raise exc
  File "/config/custom_components/peaqev/sensors/peaq_sensor.py", line 47, in async_update
    ret = await self.hub.async_request_sensor_data(
  File "/config/custom_components/peaqev/peaqservice/hub/hub.py", line 242, in async_request_sensor_data
    "is_scheduler_active": getattr(self.hours.scheduler, "scheduler_active", False),
  File "/usr/local/lib/python3.10/site-packages/peaqevcore/services/hourselection/initializers/hoursbase.py", line 41, in scheduler
    return self._scheduler
AttributeError: 'RegularHours' object has no attribute '_scheduler'
```

and
```

Logger: custom_components.peaqev.peaqservice.hub.hub
Source: custom_components/peaqev/peaqservice/hub/hub.py:150
Integration: peaqev ev-charging ([documentation](https://github.com/elden1337/hass-peaq), [issues](https://github.com/elden1337/hass-peaq/issues))
First occurred: 10:09:40 AM (1456 occurrences)
Last logged: 10:21:56 AM

Unable to handle data-update: sensor.house_power_consumption <state sensor.house_power_consumption=659; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T10:21:54.568559+02:00>|<state sensor.house_power_consumption=669; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T10:21:55.110775+02:00>. Exception: 'RegularHours' object has no attribute '_scheduler'
Unable to handle data-update: sensor.house_power_consumption <state sensor.house_power_consumption=669; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T10:21:55.110775+02:00>|<state sensor.house_power_consumption=667; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T10:21:55.646221+02:00>. Exception: 'RegularHours' object has no attribute '_scheduler'
Unable to handle data-update: sensor.peaqev_average_consumption <state sensor.peaqev_average_consumption=903; entity_id=sensor.peaqev_house_power, unit_of_measurement=W, device_class=power, icon=mdi:home-lightning-bolt, friendly_name=Peaqev Average consumption @ 2023-04-12T10:21:44.163317+02:00>|<state sensor.peaqev_average_consumption=899; entity_id=sensor.peaqev_house_power, unit_of_measurement=W, device_class=power, icon=mdi:home-lightning-bolt, friendly_name=Peaqev Average consumption @ 2023-04-12T10:21:56.166144+02:00>. Exception: 'RegularHours' object has no attribute '_scheduler'
Unable to handle data-update: sensor.house_power_consumption <state sensor.house_power_consumption=667; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T10:21:55.646221+02:00>|<state sensor.house_power_consumption=665; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T10:21:56.187139+02:00>. Exception: 'RegularHours' object has no attribute '_scheduler'
Unable to handle data-update: sensor.house_power_consumption <state sensor.house_power_consumption=665; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T10:21:56.187139+02:00>|<state sensor.house_power_consumption=662; state_class=measurement, unit_of_measurement=W, device_class=power, icon=mdi:flash-outline, friendly_name=House - Power Consumption @ 2023-04-12T10:21:56.743975+02:00>. Exception: 'RegularHours' object has no attribute '_scheduler'
```

rabakaas, 2023-04-12:
> apparently i forgot another place.
> if you have file editor access and want to test quickly please change line 242 in hub.py to this:

Thx, will try when home from work and report back. MvhRobert12. apr.

elden1337, 2023-04-12:
Ok, apparently more coding required

elden1337, 2023-04-12:
there we go https://github.com/elden1337/hass-peaq/releases/tag/v2.9.3
@kirilldem @rabakaas 
now i've tested it for a while here

closed 2023-04-12 by elden1337

kirilldem, 2023-04-12:
Seems to be running without errors in the log (no price aware right now).
I'll keep you posted.

rabakaas, 2023-04-13:
Running all night without errors here also. Thanks!

rabakaas, 2023-04-13:
Seems like Peaq is adjusting current constantly and charger is strugling to keep up with all adjustments, never seen this earlier..

[image]

elden1337, 2023-04-13:
it will update every second minute if needed to keep up with the current peak and rolling 5-min average consumption. but in this case it looks as if it is changing back directly when easee s been updated?! will keep an eye out if i experience the same thing. @rabakaas 

ps. this is a new issue if it needs fixing.

rabakaas, 2023-04-17:
I have been investigating a little, and maybe have a clue..
Watching Peaq allowed current during a charge and then comparing with logs it seems that not every change in Peaq allowed current is showing in the log. But when example custom_components.peaqev.peaqservice.chargecontroller.charger.charger] Calling charger set_charger_dynamic_limit for domain 'easee' with parameters: {'charger_id': 'EHWV5Z55', 'current': 16 is showing up, Easee charger is updating. Seems like every other change in Peaq allowed current is not showing up in log and then there is a mismatch between charger and PeaqEv. 

Charging with 10A, and PeaqEv is 10A
2023-04-17 18:00:01.562 DEBUG (MainThread) [custom_components.peaqev.peaqservice.chargecontroller.charger.charger] Calling charger set_charger_dynamic_limit for domain 'easee' with parameters: {'charger_id': 'EHWV5Z55', 'current': 10}
2023-04-17 18:00:01.565 DEBUG (MainThread) [custom_components.easee.services] Call set_current service on charger_id: EHWV5Z55
2023-04-17 18:00:01.565 DEBUG (MainThread) [custom_components.easee.services] Execute_service: set_charger_dynamic_limit {'charger_id': 'EHWV5Z55', 'current': 10}

Then this, 10A is Easee and 30A is PeaqEv
[image]

Next PeaqEv service in log calls for 32A and Easee is adjusting. Not able to find the 30A service call from Peaq in the logs:

2023-04-17 18:02:01.674 DEBUG (MainThread) [custom_components.peaqev.peaqservice.chargecontroller.charger.charger] Calling charger set_charger_dynamic_limit for domain 'easee' with parameters: {'charger_id': 'EHWV5Z55', 'current': 32}
2023-04-17 18:02:01.676 DEBUG (MainThread) [custom_components.easee.services] Call set_current service on charger_id: EHWV5Z55
2023-04-17 18:02:01.676 DEBUG (MainThread) [custom_components.easee.services] Execute_service: set_charger_dynamic_limit {'charger_id': 'EHWV5Z55', 'current': 32}
2023-04-17 18:02:01.677 DEBUG (MainThread) [pyeasee.easee] POST: /api/chargers/EHWV5Z55/settings ({'json': {'dynamicChargerCurrent': 32}})

Maybe a hopeless explanation of it, but giving it a long shot....

elden1337, 2023-04-17:
@rabakaas thanks a lot for input!
I think i have to look at the algorithm part for calculating it to not make it fluctuate so much (will do that). 
The reason why you see a missmatch is most def because:

Updatecurrent spins in it's own threaded loop, and waits for a diff in amps from peaqev vs the charger. Then it updates. 
Then it must wait for at least 120s before doing it again. I think that during this time we miss some of the changes.