elden1337/hass-peaq #265: Keba Charging Station

mattsve, 2023-05-22:
Would be nice to get support for Keba Charging station
https://www.home-assistant.io/integrations/keba/
https://github.com/home-assistant/core/tree/dev/homeassistant/components/keba

elden1337, 2023-05-23:
I have a question i'm having a hard time finding out from the repos. I need to tell my aggregator at least one of each of these states, what would the state be saying if car is not connected? @mattsve 

```python
self.chargerstates[ChargeControllerStates.Idle] = [] #??
self.chargerstates[ChargeControllerStates.Connected] = ["ready for charging"]
self.chargerstates[ChargeControllerStates.Charging] = ["charging"]
```

mattsve, 2023-05-23:
"not ready for charging". 
Just let me know if there is anything else you need to know.

elden1337, 2023-05-24:
and, nevermind the "chargerid" in the setup. Not needed for Keba .

mattsve, 2023-05-24:
Ok, first try of setup. Haven't tried charging yet, just setup. No errors during setup. This is the output in the log. The last lines for sensor.peaqev_energy_including_car and sensor.peaqev_energy_cost_integral keeps repeating, maybe it is something else with my setup? 

> 2023-05-24 16:06:41.563 WARNING (MainThread) [custom_components.peaqev.peaqservice.chargertypes.types.keba] You are initiating Keba as Chargertype. Bare in mind that this chargertype has not been signed off in testing and may be very unstable. Report findings to the developer.
> 2023-05-24 16:06:41.568 DEBUG (MainThread) [custom_components.peaqev.peaqservice.chargertypes.entitieshelper] entityschema is: keba_p30 at 1684937201.568637
> 2023-05-24 16:06:41.568 INFO (MainThread) [custom_components.peaqev.peaqservice.chargertypes.chargertype_factory] Set up charger-class for chargertype Keba is done. attempts:1
> 2023-05-24 16:06:41.570 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.nordpool.nordpool] Nordpool has been set up and is ready to be used with sensor.nordpool
> 2023-05-24 16:06:41.572 DEBUG (MainThread) [custom_components.peaqev.services] Registered services: dict_keys([<ServiceCalls.ENABLE: 'enable'>, <ServiceCalls.DISABLE: 'disable'>, <ServiceCalls.OVERRIDE_NONHOURS: 'override_nonhours'>, <ServiceCalls.SCHEDULER_SET: 'scheduler_set'>, <ServiceCalls.SCHEDULER_CANCEL: 'scheduler_cancel'>, <ServiceCalls.OVERRIDE_CHARGE_AMOUNT: 'override_charge_amount'>])
> 2023-05-24 16:06:41.638 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.nordpool.dynamic_top_price] Error in async_get_current_month: mean requires at least one data point
> 2023-05-24 16:06:41.638 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.nordpool.dynamic_top_price] Error in async_get_max: list index out of range
> 2023-05-24 16:06:41.638 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.nordpool.nordpool] (0, <AverageType.ERROR: 'Error'>)
> 2023-05-24 16:06:41.675 DEBUG (MainThread) [custom_components.peaqev.peaqservice.util.extensionmethods] Chargecontroller is initialized and ready to work.
> 2023-05-24 16:06:41.675 DEBUG (MainThread) [custom_components.peaqev.peaqservice.chargecontroller.ichargecontroller] async_check_broadcasting: ChargeControllerStates.Idle -> ChargeControllerStates.Connected
> 2023-05-24 16:06:41.686 DEBUG (MainThread) [custom_components.peaqev.sensors.money_sensor] last state of Peaqev Hour controller = None
> 2023-05-24 16:06:41.760 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.state_changes.istate_changes] Unable to update peak to unknown
> 2023-05-24 16:06:41.760 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.state_changes.istate_changes] Unable to add consumption to savings
> 2023-05-24 16:06:42.567 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.observer.observer_coordinator] ready to broadcast: daily average price changed with params: 17.8475
> 2023-05-24 16:06:42.568 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.observer.observer_coordinator] ready to broadcast: hub initialized with params: None
> 2023-05-24 16:06:43.568 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.observer.observer_coordinator] ready to broadcast: dynamic max price changed with params: 0
> 2023-05-24 16:06:44.568 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.observer.observer_coordinator] ready to broadcast: car connected with params: None
> 2023-05-24 16:06:44.568 DEBUG (MainThread) [custom_components.peaqev.peaqservice.chargecontroller.charger.savings_controller] SavingsController: async_enter
> 2023-05-24 16:06:50.705 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.max_min_controller] Resetting max charge to static value 28 because of 0
> 2023-05-24 16:06:50.708 ERROR (MainThread) [homeassistant.helpers.event] Error while processing state change for sensor.peaqev_energy_including_car
> Traceback (most recent call last):
>   File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 287, in _async_state_change_dispatcher
>     hass.async_run_hass_job(job, event)
>   File "/usr/src/homeassistant/homeassistant/core.py", line 607, in async_run_hass_job
>     hassjob.target(*args)
>   File "/usr/src/homeassistant/homeassistant/components/utility_meter/sensor.py", line 450, in async_reading
>     for sensor in self.hass.data[DATA_UTILITY][self._parent_meter][
> KeyError: 'sensor.peaqev_energy_including_car_daily'
> 2023-05-24 16:06:50.709 ERROR (MainThread) [homeassistant.helpers.event] Error while processing state change for sensor.peaqev_energy_including_car
> Traceback (most recent call last):
>   File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 287, in _async_state_change_dispatcher
>     hass.async_run_hass_job(job, event)
>   File "/usr/src/homeassistant/homeassistant/core.py", line 607, in async_run_hass_job
>     hassjob.target(*args)
>   File "/usr/src/homeassistant/homeassistant/components/utility_meter/sensor.py", line 450, in async_reading
>     for sensor in self.hass.data[DATA_UTILITY][self._parent_meter][
> KeyError: 'sensor.peaqev_energy_including_car_monthly'
> 2023-05-24 16:06:50.710 ERROR (MainThread) [homeassistant.helpers.event] Error while processing state change for sensor.peaqev_energy_including_car
> Traceback (most recent call last):
>   File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 287, in _async_state_change_dispatcher
[... cut ...]
>   File "/usr/src/homeassistant/homeassistant/helpers/event.py", line 287, in _async_state_change_dispatcher
>     hass.async_run_hass_job(job, event)
>   File "/usr/src/homeassistant/homeassistant/core.py", line 607, in async_run_hass_job
>     hassjob.target(*args)
>   File "/usr/src/homeassistant/homeassistant/components/utility_meter/sensor.py", line 450, in async_reading
>     for sensor in self.hass.data[DATA_UTILITY][self._parent_meter][
> KeyError: 'sensor.peaqev_energy_cost_integral_monthly

elden1337, 2023-05-24:
Thanks! Will look tomorrow, afk right now.

elden1337, 2023-05-24:
From reading the logs here there seem to be a couple of hickups. 
First ill try and fix is that the chargecontroller does not have a state. That indicates that keba is showing another state than i added (that where i needed help previously).  
I guess there are multiple states that i should interpret as "connected" and "charging". 

If you have insight to that let me know, otherwise ill continue checking the keba-repo tomorrow.

elden1337, 2023-05-24:
Specifically this part:

2023-05-24 16:23:57.994 DEBUG (MainThread) [custom_components.peaqev.peaqservice.chargecontroller.ichargecontroller] async_check_broadcasting: ChargeControllerStates.Connected -> ChargeControllerStates.Error

What was the state of your keba at this time? It managed to create a connected state in peaqev, but then went to error (probably from when charging begun?)

mattsve, 2023-05-24:
It must be this sensor: binary_sensor.keba_p30_charging_state
State is on if charging, otherwise off both if connected or not
It has an attribute 'status', I have seen the following values:
* not ready for charging - when the car was not plugged in
* ready for charging - when the car is plugged in
* charging - when the car is charging

Right now the car is connected but not charging since it programmed to charge 01-06. The states are the following
| Entity | State | Attributes |
|--------|------|-------------|
|binary_sensor.keba_p30_charging_state|off|status: ready for charging|
| switch.peaqev_charger_enabled|off||
| sensor.peaqev_charger_controller | Connected ||

If I enable switch.peaqev_charger_enabled 
| Entity | State | Attributes |
|--------|------|-------------|
|binary_sensor.keba_p30_charging_state|off|status: ready for charging|
| switch.peaqev_charger_enabled|on||
| sensor.peaqev_charger_controller | Error ||

elden1337, 2023-05-25:
ok, this is the way i set up the sensors for you in the state you currently have:

```python
async def async_set_sensors(self) -> None:
        self.entities.maxamps = f"sensor.{self.entities.entityschema}_Max_Current"
        self.entities.powermeter = f"sensor.{self.entities.entityschema}_Charging_Power"
        self.options.powermeter_factor = 1000
        self.entities.powerswitch = f"binary_sensor.{self.entities.entityschema}_Status"
        self.entities.ampmeter = f"sensor.{self.entities.entityschema}_Max_Current"
        self.entities.chargerentity = f"binary_sensor.{self.entities.entityschema}_Charging_State|status"
```

the issue is probably in either the ...powerswitch or the ...chargerentity as you also mention. 
the later is using the attribute "status", and that one managed to set the connected state before error, i i'm lingering towards the powerswitch. 
The thing with this is that in other brands, like Chargeamps, there is a switch within Home Assistant that turns charging on/off (so i'm checking that one for state as fallback). I guess i'll have to work through the code to make it optional and probably remove this one from Keba since that binary sensor i'm using will always be on if you are online.

mattsve, 2023-05-25:
How do you turn charger on/off?
If I use the service keba.disable, the status is as follows:
| Entity | State | Attributes |
|--------|------|-------------|
|binary_sensor.keba_p30_charging_state|off|status: authorization rejected|

Regarding Max_Current, it reported 63A, but it is configured on the unit via dip-switches to max 10A I've experimented with the service keba.set_current.
Max_Current: 63 (cannot return to this after I changed current to something else, it keeps the old value)
| Entity | State | Attributes |
|--------|------|-------------|
|sensor.keba_p30_max_current|63|max_current_hardware: 10|
|binary_sensor.keba_p30_charging_state|off|max_charging_rate: 10|

Max_Current: 12
| Entity | State | Attributes |
|--------|------|-------------|
|sensor.keba_p30_max_current|12|max_current_hardware: 10|
|binary_sensor.keba_p30_charging_state|off|max_charging_rate: 10|

Max_Current: 10
| Entity | State | Attributes |
|--------|------|-------------|
|sensor.keba_p30_max_current|10|max_current_hardware: 10|
|binary_sensor.keba_p30_charging_state|off|max_charging_rate: 10|

Max_Current: 6
| Entity | State | Attributes |
|--------|------|-------------|
|sensor.keba_p30_max_current|6|max_current_hardware: 10|
|binary_sensor.keba_p30_charging_state|off|max_charging_rate: 6|

elden1337, 2023-05-25:
i use the servicecalls for keba:

```python
@property
    def call_on(self) -> CallType:
        return CallType(
            "enable",
            {},
        )

    @property
    def call_off(self) -> CallType:
        return CallType(
            "disable",
            {},
        )

    @property
    def call_resume(self) -> CallType:
        return self.call_on

    @property
    def call_pause(self) -> CallType:
        return self.call_off

    @property
    def call_update_current(self) -> CallType:
        return CallType(
            "set_current",
            {CURRENT: "current"},
        )
```

didn't find specific pause/resume so utilizing on/off there as well as for chargeamps for example.

We'll get to that part eventually (the updatecurrent), but for now i need to see why it won't give me a state properly. I'm clearly looking at the wrong set of sensors. pushing something later for you to pull.

mattsve, 2023-05-25:
The charging station has the possibility to use RFID tags, and you "authorize" charging for a tag. So I guess that's where the language comes from. The UDP protocol the integration uses has no authorization or credentials.

elden1337, 2023-05-25:
please pull latest from this branch and try when you have time. 
i stupidly added the sensor-schema with capitalized so i'm quite sure peaqev never even found your sensors but did not fail bc of bad coding in that setup part from me.

mattsve, 2023-05-25:
Setup went fine again, produced these logs, charger was in disabled state:
> 2023-05-25 21:02:30.410 WARNING (MainThread) [custom_components.peaqev.peaqservice.chargertypes.types.keba] You are initiating Keba as Chargertype. Bare in mind that this chargertype has not been signed off in testing and may be very unstable. Report findings to the developer.
> 2023-05-25 21:02:30.416 DEBUG (MainThread) [custom_components.peaqev.peaqservice.chargertypes.entitieshelper] entityschema is: keba_p30 at 1685041350.416203
> 2023-05-25 21:02:30.416 INFO (MainThread) [custom_components.peaqev.peaqservice.chargertypes.chargertype_factory] Set up charger-class for chargertype Keba is done. attempts:1
> 2023-05-25 21:02:30.420 DEBUG (MainThread) [custom_components.peaqev.services] Registered services: dict_keys([<ServiceCalls.ENABLE: 'enable'>, <ServiceCalls.DISABLE: 'disable'>, <ServiceCalls.OVERRIDE_NONHOURS: 'override_nonhours'>, <ServiceCalls.SCHEDULER_SET: 'scheduler_set'>, <ServiceCalls.SCHEDULER_CANCEL: 'scheduler_cancel'>, <ServiceCalls.OVERRIDE_CHARGE_AMOUNT: 'override_charge_amount'>])
> 2023-05-25 21:02:30.489 DEBUG (MainThread) [custom_components.peaqev.peaqservice.util.extensionmethods] Chargecontroller is initialized and ready to work.
> 2023-05-25 21:02:30.489 DEBUG (MainThread) [custom_components.peaqev.peaqservice.chargecontroller.ichargecontroller] async_check_broadcasting: ChargeControllerStates.Idle -> ChargeControllerStates.Connected
> 2023-05-25 21:02:30.492 DEBUG (MainThread) [custom_components.peaqev.sensors.session_sensor] last state of Peaqev Session energy = <state sensor.peaqev_session_energy=0.0; average_session=0.0, remaining charge=30.0, unit_of_measurement=kWh, device_class=energy, icon=mdi:ev-station, friendly_name=Peaqev Session energy @ 2023-05-24T16:06:53.768340+02:00>
> 2023-05-25 21:02:30.493 DEBUG (MainThread) [custom_components.peaqev.sensors.sql_sensor] last state of Peaqev peak = <state sensor.peaqev_peak=4.8; observed_peak=2.28, peaks_dictionary=m=5, p=24h22=2.28, 25h0=7.21, unit_of_measurement=kWh, device_class=energy, icon=mdi:chart-donut-variant, friendly_name=Peaqev peak @ 2023-05-25T00:59:53.941149+02:00>
> 2023-05-25 21:02:30.526 ERROR (MainThread) [homeassistant.components.filter.sensor] Could not convert state:  (<class 'str'>) to number
> 2023-05-25 21:02:30.528 ERROR (MainThread) [homeassistant.components.filter.sensor] Could not convert state:  (<class 'str'>) to number
> 2023-05-25 21:02:35.839 ERROR (MainThread) [homeassistant.components.filter.sensor] Could not convert state:  (<class 'str'>) to number
> 2023-05-25 21:02:35.840 ERROR (MainThread) [homeassistant.components.filter.sensor] Could not convert state:  (<class 'str'>) to number
> 2023-05-25 21:02:35.850 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.observer.observer_coordinator] ready to broadcast: hub initialized with params: None
> 2023-05-25 21:02:35.871 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.observer.observer_coordinator] ready to broadcast: update charger enabled with params: False
> 2023-05-25 21:02:36.846 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.observer.observer_coordinator] ready to broadcast: car connected with params: None
> 2023-05-25 21:02:36.846 DEBUG (MainThread) [custom_components.peaqev.peaqservice.chargecontroller.charger.savings_controller] SavingsController: async_enter
> 2023-05-25 21:02:37.847 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.observer.observer_coordinator] ready to broadcast: update latest charger start with params: None

I didn't touch the system afterwards, 20 minutes later it kept repeating this

> 2023-05-25 21:24:24.167 ERROR (MainThread) [homeassistant.helpers.entity] Update for sensor.peaqev_power_canary_allowed_amps_3_phase fails
> Traceback (most recent call last):
>   File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 554, in async_update_ha_state
>     await self.async_device_update()
>   File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 779, in async_device_update
>     await self.async_update()
>   File "/config/custom_components/peaqev/sensors/power/powercanary_sensor.py", line 115, in async_update
>     self._state = max(ret.values())
> ValueError: max() arg is an empty sequence

When I toggled switch.peaqev_charger_enabled to on the following was logged, the last line keeps on repeating. Nothing changes when I manually enable the charger:
> 2023-05-25 22:04:16.932 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.observer.observer_coordinator] ready to broadcast: update charger enabled with params: True
> 2023-05-25 22:04:17.932 DEBUG (MainThread) [custom_components.peaqev.peaqservice.hub.observer.observer_coordinator] ready to broadcast: update latest charger start with params: None
> 2023-05-25 22:04:20.769 DEBUG (MainThread) [custom_components.peaqev.peaqservice.chargecontroller.ichargecontroller] async_check_broadcasting: ChargeControllerStates.Connected -> ChargeControllerStates.Error
> 2023-05-25 22:04:20.769 DEBUG (MainThread) [custom_components.peaqev.peaqservice.chargecontroller.charger.charger] Could not match any chargecontroller-state. state: ChargeControllerStates.Error

As soon as switch.peaqev_charger_enabled is toggled to off, it goes back to Connected state. The car is charging at that point.
binary_sensor.keba_p30_charging_state.status is 'charging'

elden1337, 2023-05-25:
Thanks. 

This is the tennis game we'll have to play i'm afraid. But im used to it by now, easee was a mess to figure out. 
I'll continue...

elden1337, 2023-05-27:
getting any further with the latest checkins? trying to ensure case-invariance on the states as that's something i realized might crap it out. 
@mattsve

mattsve, 2023-05-27:
More or less the same result as before, the only new thing in the logs is this that keeps repeating:
> [peaqevcore.models.hub.hubmember] could not set value for sensor.keba_p30_max_current to None of type <class 'int'> because of float() argument must be a string or a real number, not 'NoneType'

I did some digging in the code to see if I can help (disclaimer, I do not know python or home assistants api) and added debug output to see if I could see why it went to Error-state.
On line 148 in ichargecontroller.py the _state returned is 'off' but it seems that the code expects the value that is in the attribute status on binary_sensor.keba_p30_charging_state. Hope this helps.

 https://github.com/elden1337/hass-peaq/blob/dcc9f09ca6c6a042c9c4019214586fe75968d812/custom_components/peaqev/peaqservice/chargecontroller/ichargecontroller.py#L148

mattsve, 2023-06-01:
I found out today that if you use the service keba.stop and a car is not connected the status attribute stays at "not ready for charging", if you then plug in the car it still says "not ready for charging". It seems that that is not a reliable way of telling if the car is plugged in or not, or if the charging station is started or stopped. I think binary_sensor.keba_p30_plug is the way of knowing if the car is plugged in or not.

I will try the beta version of the integration, there seems to be more sensors and fewer attributes. https://github.com/dannerph/homeassistant-keba

mattsve, 2023-06-01:
The beta version has a sensor.keba_p30_status, seems to have the same strings as the status attribute in the old integration.
Will try tomorrow what the statuses says if the charger is disabled and the car is not connected

mattsve, 2023-06-02:
If the car is not connected and the charger is disabled Status stays at 'not ready for charging'. But the sensor binary_sensor.keba_p30_enable_user goes to off when the charger is disabled.

elden1337, 2023-06-02:
Thanks. Away today but maybe tomorrow.

elden1337, 2023-06-04:
i've pushed a small change to the core-repo that will force an update of the chargerobject before getting the value (thus handling the attribute-problem i hope).

mattsve, 2023-06-06:
Sorry, still the same problem going to error state when enabling peaqev.
My guess is that the beta would be easier to integrate. There is a status sensor instead of an attribute and there is a sensor indicating if the charging station is enabled or disabled.

elden1337, 2023-06-06:
;( ok. It sucks that i cannot have them all here and test directly.

mattsve, 2023-06-06:
According to your profile you are based in Gothenburg? You are welcome to visit anytime if it would help :)

elden1337, 2023-06-07:
Working simultaneosly with https://github.com/elden1337/hass-peaq/pull/267 for Garo. That one's being developed in develop-branch, and i merged some changes to "your" branch now to accommodate better update of sensor-data that comes from attributes.

Not verified to work better with the Garo yet, but jut fyi. It might help things come further.

mattsve, 2023-06-10:
Sorry, no chage :( "Could not match any chargecontroller-state. state: ChargeControllerStates.Error"

elden1337, 2023-06-10:
Ok. Gonna continue with the garo in another branch and see what's happening there.

closed 2024-10-26 by elden1337