elden1337/hass-peaq #296: Peaq without controlling the actual charger

reinestenberg, 2023-09-29:
Feature:
Allow Peaq to run without a specific charger.
In this case Peaq outputs the allowed current to a sensor value instead of controlling the charger.
This should be chosen in the configuration flow.
This sensor value can then be used by to control multiple chargers of same or different type or to control a combination of chargers and heating appliances/dishwashers/washing machines/dryers/whatever.

Why?
I think that ev charging is dependent on a lot of factors which is very different for each user and thus impossible to incorporate into one application. In my case it is normally sufficient to charge the car a little more than once per week for normal commuting, but if the car is to be used for a longer trip during the weekend it might need to be at 100% on friday afternoon. 
For another person, the car might need to be at at least 60% each morning.

elden1337, 2023-10-02:
interesting, and probably not a super hefty task to add. 
But, if you do have any of the supported chargers (which hopefully work), perhaps we should have a refactoring look at the scheduler service as well to make sure it works. 

The idea behind that is just what you mention in the later part of the request here, that you can, through automations or manually set a needed amount by time x.

reinestenberg, 2023-10-04:
I was not aware of the "no charger" option. I don't think it was present when I first installed peaq.
Now I tested it:
Update to latest peaq
reboot
Delete current hub using easee
Create new hub: regular, sensor was validated, including car, no charger, blank charger id (should this even be asked if "no charger"?)

Peaq integration shows red exclamation mark: Failed to set up
Log:
AttributeError: 'HubSensors' object has no attribute 'amp_meter'
line 172 in async_setup
self.amp_meter

I'll try to post the complete log entry but it is on the phone

reinestenberg, 2023-10-04:
This error originated from a custom integration.
Logger: homeassistant.config_entriesSource: custom_components/peaqev/peaqservice/hub/sensors/ihub_sensors.py:172Integration: peaqev ev-charging (documentation, issues)First occurred: 07:54:25 (4 occurrences)Last logged: 08:07:41
Error setting up entry sensor.pws_power_sum for peaqevTraceback (most recent call last):  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 387, in async_setup    result = await component.async_setup_entry(hass, self)             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^  File "/config/custom_components/peaqev/__init__.py", line 30, in async_setup_entry    await hub.setup()  File "/config/custom_components/peaqev/peaqservice/hub/hub.py", line 92, in setup    self.sensors: HubSensors = await HubSensorsFactory.async_create(hub=self)                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^  File "/config/custom_components/peaqev/peaqservice/hub/sensors/hubsensors_factory.py", line 16, in async_create    return await HubSensorsFactory.async_setup(hub, sensors())           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^  File "/config/custom_components/peaqev/peaqservice/hub/sensors/hubsensors_factory.py", line 20, in async_setup    await sensors.async_setup(  File "/config/custom_components/peaqev/peaqservice/hub/sensors/ihub_sensors.py", line 172, in async_setup    self.amp_meter,    ^^^^^^^^^^^^^^AttributeError: 'HubSensors' object has no attribute 'amp_meter'

reinestenberg, 2023-10-04:
> interesting, and probably not a super hefty task to add. But, if you do have any of the supported chargers (which hopefully work), perhaps we should have a refactoring look at the scheduler service [...]

What I have today is:
A max price helper that sets the maximum charge price.
A minimum charge level (%) that the car must reach latest at 6 on weekdays and 10 on weekends
Possibility to set max charge level (%) on the car
A "charge now" toggle for overriding everything

Using this and the current SOC (%) in the car I calculate the number of hours needed and picks the cheapest ones.
I then output a per phase max current (also considering the fuse rating) to the charger network (essentially the one charger I have, but would affect all easee chargers should I have hade more than one) using a different service than Peaq.

This works well, but not together with peaq (at least not older versions). I have on several occasions not been able to charge and one of them was really important to me. What happened on these occasions were that the max charger current set by peaq was forced to 0 (even with peaq turned off with the GUI toggle slider). I still have no evidence of where this bug is/was located but the easee api/backend/charger is a strong suspect.
So the idea from my side is to keep it simple. I like the idea in linux (et al.) where you have several quite simple tools that can do wonders when combined, rather than the windows quite cumbersome advanced tools.
E.g. How is this done in windows: grep -R -l --include="*.c" uint32 | sort -r
In Home Assistant something like this could be implemented using the Node Red add-on.

Peaq is a great tool, but it suffers a bit from the "one great tool" syndrome. Perhaps this is the correct way to do it in Home Assistant. Certainly it is a good way to do it for a lot of people who don't want to get in to the details too much. But for me (and I guess a certain amount of others) it does too much, and to me it is hard to grasp exactly what the different settings/sensors actually do/show. 

So for my part, the peak-part of Peaq is what I need now, and you seem to already have fulfilled my needs :)

closed 2024-06-21 by elden1337