elden1337/hass-peaq #307: Feature request: Skånska Energi

enoch85, 2023-10-23:
Just wanted to add Skånska Energi to the mix of available net owners, if possible?

elden1337, 2023-10-24:
I cant find any effektavgift (peak energy charges) on their consumer subscriptions. I only add those for calculations of peak avoidances. 
Have they announced a coming change to the model?

enoch85, 2023-10-24:
> I cant find any effektavgift (peak energy charges) on their consumer subscriptions. I only add those for calculations of peak avoidances. Have they announced a coming change to the model?

No they don't have that and nothing that I'm aware of is coming. I thought the different energy providers list were current transfer fees, but I guess I was wrong.

elden1337, 2023-10-24:
no, this is not accounted for, that's not how the integration works. 
transfer fees can be added either as a template from nordpool or (i think) in energidatasaervice by default?

enoch85, 2023-10-24:
> no, this is not accounted for, that's not how the integration works. transfer fees can be added either as a template from nordpool or (i think) in energidatasaervice by default?

I actually created a helper for it. Just thought getting them automatically was more convenient.

Will switch to Greenly soon, and they have their own app were everything is included. So I think I will use [this integration](https://github.com/linsvensson/sensor.greenely) to fetch the actual price + my helper variables. 

I guess this can be closed.

closed 2023-10-24 by enoch85