elden1337/hass-peaq #319: Unable to specify power sensor during setup

JonathanJorstad, 2023-12-10:
I have tried specifying both the sensor entity ID and the name, but they do not seem to be recognized. I am using a the AMSleser powermeter with MQTT.

During configuration, it is unclear whether the power sensor entity ID or entity name should be specified - or device name? There is no drop down/quick search results as you type in the configuration text box. 

[image]
[image]
[image]

Error log:
Loggdetaljer (FEIL)
Logger: homeassistant.config_entries
Source: config_entries.py:402
First occurred: 12:07:42 (1 occurrences)
Last logged: 12:07:42

Error setting up entry sensor.ams_p for peaqev
Traceback (most recent call last):
  File "/config/custom_components/peaqev/peaqservice/chargertypes/chargertype_factory.py", line 62, in async_create
    ret.is_initialized = await ret.async_setup()
                         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/peaqev/peaqservice/chargertypes/icharger_type.py", line 78, in async_setup
    update_current=self.call_update_current,
                   ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/peaqev/peaqservice/chargertypes/types/zaptec.py", line 86, in call_update_current
    raise NotImplementedError
NotImplementedError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 402, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/peaqev/__init__.py", line 28, in async_setup_entry
    hub = await HubFactory.async_create(hass, options, DOMAIN)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/peaqev/peaqservice/hub/hub_factory.py", line 36, in async_create
    return await HubFactory.async_setup(hub(hass, options, domain))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/peaqev/peaqservice/hub/hub_factory.py", line 40, in async_setup
    hub.chargertype = await ChargerTypeFactory.async_create(hub.state_machine, hub.options)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/peaqev/peaqservice/chargertypes/chargertype_factory.py", line 77, in async_create
    raise Exception
Exception

elden1337, 2023-12-10:
seems your powersensor is correctly put (sensor.ams_p as your entry is called is the right way). but it seems as if the zaptec-integration handling is being angry for some reason .is this all you get from peaqev?

elden1337, 2023-12-10:
i'm going to push a quick hotfix-update for you which is hope should make it better.
please note that zaptec integration has yet to get update-current features since they aren't ready (or ready just now perhaps but i haven't had time to implement them yet)

elden1337, 2023-12-10:
there: https://github.com/elden1337/hass-peaq/releases/tag/3.3.1

hopefully it will work better for you. keeping this ticket open til you say it's ok.

JonathanJorstad, 2023-12-10:
I think it worked, at least I am now seeing the 4 devices in peaqev.

elden1337, 2023-12-10:
splendid. 
As mentioned i don't have updatecurrent for zaptec (perhaps it's ready to implement by now but noones asked for it lately). I'll have a look at it.

Happy charging!

elden1337, 2023-12-10:
hmm, looks like they've broken the whole integration. but you are able to get peaqev running without any seen issues now? @JonathanJorstad ?

closed 2023-12-10 by elden1337

JonathanJorstad, 2023-12-10:
Not really actually, it seems that the Zaptec charger is not being
controlled by peaqev. I just tried it, disabling any smart eco charging and
peak shaving within Zaptec. It charged with maximum power (7.2 kW). Prior
to this I had set limit in peaqev to 10 kW and enabled the peaqev charger.
Peaqev did not seem able to limit the charge power. I turned off the peaqev
charger controller, the charging continued regardless.

The peaqev charger controller does report an error.

elden1337, 2023-12-10:
Yepp. Got it. The breaking change from them has made it useless. I have to refactor the code. Sorry about that but I couldnt foresee that change on their part.

elden1337, 2023-12-10:
New feature for it is #320