elden1337/hass-peaq #321: Refactor Zaptec setup (breaking change on their part)

elden1337, 2023-12-11:
https://github.com/custom-components/zaptec#bangbang-breaking-change

elden1337, 2023-12-11:
@JonathanJorstad i realize this will go so much quicker if i could get a list of sensors from you that you would claim are:

- powermeter that shows how much W or kW zaptec is currently pulling
- the sensor which is the main zaptec-entity. Showing the states of your connected car, if it's charging etc.
- the sensor which is the switch to turn on/off
- the sensor that shows the current max amps delegated to zaptec.

JonathanJorstad, 2023-12-12:
Ok, there's always a risk of me making a mistake but i think it's like this:
There are three devices:
1. the installation (top entity - this is where you set the current limitation)
2. the circuit (this is mainly a passive device, just conatining the information about the current limitation in the physical electrical circuit - should not be changed)
3. and finally the charger (or multiple chargers).
This division in three levels mirrors the setup in the Zaptec native application.

- powermeter that shows how much W or kW zaptec is currently pulling
sensor.[CHARGER NAME]_charge_power
This is provided for the "bottom-device" charger, where [CHARGER NAME] must be replaced by the name of the charger, default is "ladestasjon_1", but it can be changed by the user. If there are multiple charges in the installation, the total power will be the sum of all the chargers.

- the sensor which is the main zaptec-entity. Showing the states of your connected car, if it's charging etc.
sensor.[CHARGER NAME]_charger_mode
This varies between for instance "disconnected", "Charging", "Charge done" etc. This is the status for the specific charger.

the sensor which is the switch to turn on/off
switch.[CHARGER NAME]_charging
Alternatively, use the buttons defined for stop and resume charging.

the sensor that shows the current max amps delegated to zaptec.
number.[INSTALLATION NAME]_available_current
This is where you would manipulate the current to control the power. The max current is defined for the entire installation and not per charger (can also be set for the individual charger, but it is the "wrong" way to do it).

elden1337, 2023-12-12:
ok, so there are two variables for this integration now, CHARGER_NAME and INSTALLATION_NAME. 
hmm. that calls for a bit larger refactoring i'm afraid, let me think.

gustafssone, 2024-02-12:
I upgraded the Zaptec installation and the new version with updating amps works just fine. Here is a blueprint that defines loadbalancing too, which I use: https://github.com/svenakela/ha/blob/main/zaptec/charger-balancing-blueprint.yaml . It might give you hints on what variables to use.

closed 2024-10-26 by elden1337