elden1337/hass-peaq #322: Control 2 Easee chargers?

rabakaas, 2023-12-25:
Is it possible to add a second chargers? I have added another Easee charger to my installation and want Peaqev to control both😃

elden1337, 2024-01-03:
note to self, duplicates: #153

elden1337, 2024-01-03:
@rabakaas if i would add support for multiple chargers (or ports in certain cases, like chargeamps aura), do you see that there should be a prioritization between the two?

- should one be considered primary and the other one secondary?
- should the one where there has been less charge added (if both are connected) be prioritized?
etc.

I'd like your thoughts on these and possibly other parameters since adding a secondary can be everything from just allowing multiple installations (with their own threshold then) to actually store and prioritize based on x factors.

rabakaas, 2024-01-04:
Prioritization between the two is not a problem for me, but maybe for other users. If i need to charge one car rapidly I will simply plug in just that car. 
My Easee chargers are on the same circuit with one charger as master, maybe a possible solution can be replacing dynamic_charger_limt with dynamic_circuit_limit on the master charger? And max_charger_limit with max_circuit_limit, current with circuit_current. Then only the master charger need to be controlled by Peaqev.

closed 2024-10-26 by elden1337