elden1337/hass-peaq #331: Stopped until further notice

pezlin, 2024-04-11:
Quite often I see that the status of peaqev is Stopped until further notice. It can be at any time of the day. A restart of HA is the only way to solve it that I have found. A reload of the integration does not solve it. Today I noticed the status after plugging in the charger cable in the car at around 16. How to troubleshoot?

pezlin, 2024-04-12:
I have the setting to always charge below 60 öre and the price yesterday was lower at that point and forward. Have I set the setting för always charge incorrectly perhaps?[image]

elden1337, 2024-04-12:
Sounds like a bug. Im gonna do a test case and see what i get, hopefully during the weekend? Which price area do you live in so i can match the correct prices?

elden1337, 2024-04-14:
Havent found any obvious bug yet. Continue looking. 
Next time this happens please comment here so that i know that day's prices. 
Also test changing your cautiontype to Suave and see if it helps.

elden1337, 2024-06-21:
Closing this as i haven't found any other reason in the code. Please reopen with new info if necessary.

closed 2024-06-21 by elden1337