elden1337/hass-peaq #347: Configuration fails at setup of price aware charging

JonathanJorstad, 2024-07-04:
After updating to 3.6.1 my previous setup/hub did not work, so I tried to configure a new one, i.e. add peaqev from scratch. I manage to specify the power sensor, selecting charger brand (Zaptec) and grid company (Glitre). But when I do not want price aware charing in the next step, I get the error "unknown error occurred" whatever I specify. I have not tried to add a price service. I have re-downloaded peaqev in hacs and restarted HA as promted.
[image]

elden1337, 2024-07-07:
If you are able to dump logs here it would help error search.

JonathanJorstad, 2024-07-07:
Since the setup fails, the integration is not listed under Settings > Devices & Services > Integrations, so it is not possible to enable debug logging. However, I attach the HA system log
[home-assistant_2024-07-07T15-00-46.102Z.log](https://github.com/user-attachments/files/16119770/home-assistant_2024-07-07T15-00-46.102Z.log)

elden1337, 2024-07-07:
think i caught it by the help of a another stranded user. pushing a fix soon which hopefully will help. otherwise we'll iterate further on it.

elden1337, 2024-07-07:
https://github.com/elden1337/hass-peaq/releases/tag/3.6.2

Please close this issue if problem goes away, otherwise comment what you get as error.

JonathanJorstad, 2024-07-07:
Confirmed working

closed 2024-07-08 by elden1337