elden1337/hass-peaq #358: Scheduler issue

kirilldem, 2024-08-27:
Just wanted to check that I am doing peaq schedule right. I attach a screenshot of what I see prior to setting any kind of scheduling, 
[image]

then I send
```
action: peaqev.scheduler_set
data:
  charge_amount: 1.36
  departure_time: 2024-08-28 08:00 
```
[image]

and it wants to schedule charging already at 2300. I am in SE3, so would expect it to charge at 0200 when the price is lowest. My peak this month is at 3kWh. Max charging per session set in peqev configuration is at 20kwh. I also attach my energy usage for the past 12 hours as I am not sure that 10.4 kwh number is right either. Given my 3 kwh peak, I should have quite a bandwidth to charge the car during the night. 

[image]

Also, I couldn't find a way to wiki, but maybe you could update https://github.com/elden1337/hass-peaq/wiki/Scheduler to:

```
action: peaqev.scheduler_set
data:
  charge_amount: 7.2
  departure_time: 2022-06-02 07:00
  schedule_starttime: 2022-06-01 18:00 #OPTIONAL
```

elden1337, 2024-08-27:
10.4 is as correct as can be im afraid (cant do advanced splitting before i move to cloud) since it uses your average 24hr consumption as base.

I ll check the schedule next time I have hack-time.

elden1337, 2024-08-28:
Checked quickly and was unable to reproduce. Think it could be something in the service where i dont handle such low prices that well, so need to look deeper.

elden1337, 2024-08-29:
I have now written two tests to try and mimick your issue, but not managing to get the wrong hours. If you can provide me with what your average 24h consumption was at the time, and also your current observed peak it would be awesome to see if i get different results. 

Both these pass:

```python
P240827CENT = [-1.392, -1.529, -1.164, -0.913, -0.057, 0.103, 2.226, 4.200, 5.627, 5.227, 5.033, 4.679,
                   4.451, 1.347, 2.602, 3.515, 4.109, 4.771, 5.227, 4.805, 3.778, 2.876, 1.267, 0.000]
P240828CENT = [-0.125, -0.638, -0.911, -0.672, -0.114, 0.000, 1.856, 4.315, 5.043, 5.180, 4.929, 4.770,
               4.485, 4.668, 4.576, 4.554, 5.658, 8.641, 9.210, 8.732, 7.252, 5.692, 4.315, 2.857]

@pytest.mark.asyncio
async def test_scheduler_correct_hours_negative_prices_cent_and_kr():
    start_dt = datetime.strptime("24-08-27 21:07", "%y-%m-%d %H:%M")
    dep_time_dt = datetime.strptime("24-08-28 08:00", "%y-%m-%d %H:%M")
    charge = 1.36
    avg_24 = 800
    peak = 1.9
    model = UpdateSchedulerDTO(
        moving_avg24=avg_24,
        peak=peak,
        chargecontroller_state=ChargeControllerStates.Stop,
        charged_amount=0,
        prices=P240827CENT,
        prices_tomorrow=P240828CENT
    )

    """Test with cents"""
    s = Scheduler(test=True)

    await s.async_create(desired_charge=charge, departuretime=dep_time_dt, starttime=start_dt)
    await s.async_update(dto=model,mockdt=start_dt + timedelta(seconds=2))

    """Test with full kr"""
    model.prices = [x / 100 for x in P240827CENT]
    model.prices_tomorrow = [x / 100 for x in P240828CENT]

    s2 = Scheduler(test=True)

    await s2.async_create(desired_charge=charge, departuretime=dep_time_dt, starttime=start_dt)
    await s2.async_update(dto=model,mockdt=start_dt + timedelta(seconds=2))

    """Assert"""
    assert s2.model.hours_charge == s.model.hours_charge
    assert s2.model.hours_charge == {
        datetime(2024, 8, 28, 2, 0): 1,
        datetime(2024, 8, 28, 3, 0): 0.2,
    }
    assert s.active is True
    assert s2.active is True
```

kirilldem, 2024-08-29:
Not sure if this is what you requested - but this is my consumption just prior to the night in question:

[image]

And my peaks were 3.1, 3.1, 3.7

I also remembered that whenever I would set peaqev schedule, the car would not normally charge fully, ending charging at around 90 percent, sometimes it was 93, another time 98. This was just random when it still had time left to charge at the cheap hours. 

So I have now set another schedule and here are the before and after results:

```
action: peaqev.scheduler_set
data:
  charge_amount: "8.58"
  departure_time: "2024-08-30 08:00"
```
[image]

[image]

elden1337, 2024-08-29:
Thanks. Will update tests and see what happens. 
But for 24 consumption i meant this sensor: sensor.peaqev_average_consumption_24h
What was its state (approx) by the time you set the scheduler yesterday?

elden1337, 2024-08-29:
Alot has been fixed in scheduler last months, so hopefully it should not stop at random.

kirilldem, 2024-08-29:
> Thanks. Will update tests and see what happens. But for 24 consumption i meant this sensor: sensor.peaqev_average_consumption_24h What was its state (approx) by the time you set the scheduler yesterday?

[image]

kirilldem, 2024-08-29:
> Alot has been fixed in scheduler last months, so hopefully it should not stop at random.

I will continue monitoring, and let you know if I see it again.

kirilldem, 2024-08-30:
> Not sure if this is what you requested - but this is my consumption just prior to the night in question: [...]

So an update on what happened tonight with the above settings. The car is charged to 92% (Consistent with what I have seen before when using the scheduler). The charger reports 5.6 kwh delivered to the car

The peaqev controller changed to _Stopped until 11:00_ at 04 in the morning (despite the original schedule saying that it would charge until 00:00 the following day).

There was only one error in the log:
```

This error originated from a custom integration.

Logger: custom_components.peaqev.peaqservice.util.extensionmethods
Source: custom_components/peaqev/peaqservice/util/extensionmethods.py:28
integration: peaqev ev-charging (documentation, issues)
First occurred: August 25, 2024 at 2:58:51 PM (9 occurrences)
Last logged: 3:00:05 AM

Fail when trying to stop connected charger. Retrying stop-attempt...
```

The charging curve looked like this:

[image]

It didn't use the high peak from the same day of 3.5. And If I look at the observed peaks, I understand why ;)

[image]

It has not updated the peaks. this is what mine look like for the month (from my smart meter):

[image]

As a data point, I sent a service call to update peaks, a couple of weeks ago, and they were updated:

[image]

kirilldem, 2024-08-30:
I had a look at the schedule screenshot again, and it shows Max charge amount of 5.2 which kinda matches with what the charger actually delivered. But this is strange since in the scheduler I requested 8.58 kwh.

elden1337, 2024-08-30:
might be the dynamism of it not playing well. as it plans a charge session it will check how much it can charge per hour based on the 24h average consumption and the peak. that will play a part in how it selects hours (how many it will need). however, as it then goes along, or also before it starts, it may update itself and change the hours because of a change in peak and/or the average consumption so that it now needs more hours (probably more expensive ones) or less (cheaper) because of higher peak or lower other consumption.

I understand that this can be complex but I have figured it's the only way we can do this in a correct way and not like the (imo) less refined commercial alternatives.
But, it also ofc means more possible issues with it. 

I'll see if i can log something differently so that I can see this behavior appear.

kirilldem, 2024-09-26:
Trying again tonight:

Peaqev proposes to charge after midnight, sounds about right:

[image]

```
action: peaqev.scheduler_set
data:
  charge_amount: "8.0"
  departure_time: "2024-09-27 08:00"
```
I then tell it that I need to charge 8.0 kwh so should take around 3 hours, and it wants to start charging already at 2100, and ideally it would start charging after 2 in the morning when its the cheapest :)

Peaqev Average consumption 24h - 724.0 W

[image]